imbushuo/mac-precision-touchpad #415: Three Finger Drag

oscarat50, 2021-07-05:
**Pre-report checks**
- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans? Yes, I did. Found no request similar to mine. 

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...] I just use three finger drag a lot, I miss that feature, it's essential to my workflow.

**Describe the solution you'd like**
A clear and concise description of what you want to happen. Implementation of the three finger drag feature.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered. None

**Additional context**
Add any other context or screenshots about the feature request here. Thanks a lot for your hard work!

MagicUtilities, 2021-07-07:
It's unlikely that the 3 finger drag feature comes to this driver because of the very nature this driver works. It only converts the raw trackpad touches so the Windows HID class driver can understand them.

It does NOT interpret the touches. As the 3 finger dragging is not part of Windows gestures, this driver would need to modify the touches on the way up (3 finger touch into 1 finger click and hold) to achieve your feature request.

See also #84 

You might consider the [Magic Utilities](https://magicutilities.net/magic-trackpad/features) which supports 3 finger dragging.

closed 2021-07-08 by imbushuo

abulka, 2021-07-27:
> You might consider the Magic Utilities which supports 3 finger dragging.

Does that paid product supplant or replace this wonderful free `mac-precision-touchpad` driver?

P.S. The three finger drag to move windows and select text is so ingrained in me from using my Mac, I miss it under Windows.

MagicUtilities, 2021-07-27:
Unfortunately by their very nature, both drivers can't coexist together so the Magic Utilities replaces this driver.

Three finger dragging is not part of the Magic Trackpad. It is only done by interpreting the raw touches by a built-in software driver on macOS. The Magic Utilities do the same on Windows.