imbushuo/mac-precision-touchpad #420: Release 3979 - Rightclick/Multitouch/Gestures not working on Bluetooth

GilbertChan, 2021-07-23:
Other than mouse moving, I am unable to right click, use multitouch and gestures when connected through Bluetooth.

Occurs when I installed Release 3979 and paired with Bluetooth on Windows 10 21H1

I'm not sure if this is expected behavior as the features have not been implemented yet or there is something wrong with my setup. These features are available when I am connected via USB on the same release.

lloydo3000, 2021-07-23:
Works for me on Bluetooth

GilbertChan, 2021-07-24:
This is what I see in my Device Manager when connected via bluetooth with the latest release installed. 
[image]

This is what I see when connected with USB.
[image]

GilbertChan, 2021-07-24:
I found the issue, I had to uninstall the Apple Bluetooth compatibility drivers. It is now working and running with the Release 3979 drivers.

closed 2021-07-24 by GilbertChan

razvan-tudosa, 2022-06-08:
Hey @GilbertChan I have the same issue, where did you find the Apple Bluetooth compatibility drivers?

GilbertChan, 2022-06-08:
@razvan-tudosa If I remember correctly, I uninstalled the Apple Bluetooth compatibility drivers in Device Manager under Human Interface Devices.

razvan-tudosa, 2022-06-08:
I see, well I did not have them in there, but I saw that I can install the drivers from the terminal as well using _chocolatey_ and after that everything worked out fine. I think drivers were not installed fine with the traditional way for some reason.

Thanks for answering anyway!