imbushuo/mac-precision-touchpad #423: suddenly stop working (freeze) until turn off/on the device again

csharp33, 2021-07-30:
Thank you for the driver that many people are able to enjoy Magic Trackpad 2 on Win10.

I searched similar issues and found #65, #268, #384, #402 but they seem to report that stopping for a while not until resetting the device so I report separately.

**Describe the bug**
While I use the trackpad connected via usbA-lightening cable to my notebook
it intermittently and suddenly stops working (freeze) and cannot move thereafter until I turn off and on the device again.

**To Reproduce**
1) turn on the trackpad
2) move around and click
3) suddenly it stops (cannot anticipate when but very frequently, saying more than 5 times in 10 minutes, and while I write this report)

**Expected behavior**
It should not stop working and at least expected to be freed after some moments.

**Traces**
[image]

I followed the instructions on #62 and noticed that I need some .pdb file to be updated, but I cannot find the .pdb file in the downloaded binaries and even source codes.

Anyhow, I traced somethings in between before and after the stoppages.

[Bring-up_csharp33.zip](https://github.com/imbushuo/mac-precision-touchpad/files/6907054/Bring-up_csharp33.zip)

**Environment**
Notebook computer (11th Gen Inttel(R) Core i7-116G7 1.20GHz, 16GB)
Windows 10 Home 64bit, 20H2 (Build: 19042.1110)
Notebook USB type-A 3.1 Gen1 to Lightning cable to the trackpad

Thank you again and hope for the problem to disappear..

Best regards,

Chanseok Park

csharp33, 2021-07-31:
Maybe it's not the problem of the magic trackpad2 and driver but of Dell WD19 dock.. I found similar problems and try to find solutions. Thank you!

imbushuo, 2021-08-01:
Doesn't sound like my issue, try to connect it directly to the machine and see if you can reproduce the bug.

csharp33, 2021-08-01:
You're right. I ordered another dock (but thunderbolt version not usb-c) and I will have usb-c dock be replaced. Anyhow, I tried previous version of your driver and there was no difference. Maybe it seems the problem of the docking station although anything else work properly I understand that magic trackpad2 is not for windows systems.. I will update when I got a new dock. Thank you!

csharp33, 2021-08-03:
I got a thunderbolt-version dock and the device work perfectly via USB. Actually, the previous one with USB-C was also ok with Bluetooth (not with USB) but then why am I need a dock? :) Anyhow, it seems that the previous dock is defective as similar symptom was found on the internet or because of bandwidth between USB-C and thunderbolt3 (this is just my guessing). Thank you and now I enjoy magic trackpad2 on Win10. Have a nice day everyday! 👍

csharp33, 2021-08-03:
I'm finding how to change the label 'bug' as it's wrong...^^;

closed 2021-08-03 by csharp33