imbushuo/mac-precision-touchpad #424: Magic Mouse driver?

Tvardovsky, 2021-08-01:
I'm using Windows 10 via Bootcamp on my Macbook Pro 15 Retina Late 2013 (MacBookPro11,3).
After installing your driver I received proper trackpad experience on the internal trackpad. Smooth scrolling (with inertia).
But I also use Magic Mouse (A1296) via Bluetooth, and now I'm wondering if any chance to implement that great scrolling with inertia from your driver to Magic Mouse too? Because now it's great on Trackpad but painfully bad on Magic Mouse (with Bootcamp drivers).
Thanks a lot for your work!

imbushuo, 2021-08-02:
As discussed in previous question, it will be too awkward to implement a magic mouse driver due to the nature of its report descriptor/report content.

closed 2021-08-02 by imbushuo

fuweichin, 2025-05-23:
Even having MacBook Pro (M3), I still use MacBook Pro (2019) to run Windows 10 for cross-platform app development for testing.

Why do I need this [mac-precision-touchpad](https://github.com/imbushuo/mac-precision-touchpad) while recent BootCamp trackpad driver already supports Windows Precision Touchpad protocol for MacBook internal trackpad and Magic Trackpad 2?

It's a pity that BootCamp mouse driver doesn't support smooth scrolling for Magic Mouse (afaik, only two mouse models, Magic Mouse on macOS and Surface Precision Mouse on Windows, natively support smooth scrolling).

Basic idea to achieve smooth scrolling for Magic Mouse on Windows:

1. Reverse-engineering Surface Precision Mouse's HID report descriptor and HID packet chracteristics
2. [Write a HID source driver by using Virtual HID Framework](https://learn.microsoft.com/en-us/windows-hardware/drivers/hid/virtual-hid-framework--vhf-) (used for scrolling)
3. [Creating a New Filter Driver](https://learn.microsoft.com/en-us/windows-hardware/drivers/develop/creating-a-new-filter-driver) for Magic Mouse converting touch input to scrolling message and pipe to the Virtual HID driver

See also [windows driver docs](https://github.com/MicrosoftDocs/windows-driver-docs/blob/staging/windows-driver-docs-pr/develop/creating-a-new-filter-driver.md)

No matter open-source or commercial, as long as they are cost-effective.

+ [Magic Utilities](https://magicutilities.net/magic-mouse/features) can enable smooth scrolling for Magic Mouse 2 BUT it's subscription-based and over-priced (16.99US$/year).
+ To disable Turbo Boost on Intel-based Mac, I spent 12.99US$ to buy [Turbo Boost Switcher Pro](https://tbswitcher.rugarciap.com/) license key
+ To use Launchpad-equivalent on Windows, I spent 76.81CN￥ to buy XLaunchpad Pro license key