imbushuo/mac-precision-touchpad #426: Bluetooth stopped working

neeko0222, 2021-08-03:
**Describe the bug**
Mousepad no longer works via bluetooth. However it can be found and be connected to Magic Mouse Pad 2 Input on bluetooth device list. 
Currently I have it plugged in via usb but when I try to reinstall the latest driver the pad stops working but can still connect to bluetooth.

I have a driver called Apple Multi-touch Trackpad HID Filter getting a This device cannot start. (Code 10) STATUS_DEVICE_POWER_FAILURE. 

Here is my driver list: https://i.imgur.com/4fnPgW5.png

**Environment**
Magic Mouse Pad 2, Driver Type USB and OS 19043.1110
I uninstalled the latest window updates since it was working correctly a couple days ago.

The only program I installed was reWASD.

neeko0222, 2021-08-04:
so i found out reWASD was causing the problem. Anyway for them to play nice?

AeliusSaionji, 2021-10-06:
Also have rewasd and getting the same error

Rokory, 2021-12-06:
I do not have re-wasd, but i have the same issue. Disabling Bluetooth and re-enabling fixes the problem, but it re-appears after some time.

Rokory, 2021-12-06:
I had the same problem with Magic Utilities. There, I fixed it checking the Keep Blutooth connection alive setting.
[image]

JoeTuohy, 2022-01-04:
I'm also facing this issue since installing reWASD. I contacted reWASD support and they've said they reached out to the developers of mac-precision-touchpad but haven't heard back yet.
I tried testing Rokory's suggestion of disabling bluetooth and re-enabling, which didn't fix it even temporarily for me. Unfortunately mac-precision-touchpad doesn't have any config settings like Keep Blutooth connection alive as far as I can tell

AeliusSaionji, 2022-01-05:
I believe rokory had an entirely different problem.

There is a definite compatibility issue- that much is clear.
I spoke with reWASD too. Their dev team have access to a magic trackpad and they have observed the issue as well.

Hope to see this resolved!

JoeTuohy, 2022-01-05:
> I believe rokory had an entirely different problem.

Yeah I also believe they have an entirely different problem, but since it was also driver related I thought testing the workaround was worth a try

> There is a definite compatibility issue- that much is clear. I spoke with reWASD too. Their dev team have access to a magic trackpad and they have observed the issue as well.
> 
> Hope to see this resolved!

That's great to hear. They told me they would consider adding Apple trackpad direct support if enough demand exists so hopefully they have enough demand!

Rokory, 2022-01-05:
Just to be clear:

Bluetooth connection works for me for some time. But after hours or minutes (very varying), it simply stops working. I use a Bluetooth stick in a desktop PC. When it stops working, I plug the BT stick out and in again. Then, it works for some time again.

Does Bluetooth work at all in your cases? If the problem is that different, I will open a new issue.

JoeTuohy, 2022-01-05:
Hi Rokory,

No, the trackpad doesn't work at all over bluetooth. The PC can connect to the trackpad over bluetooth, but no inputs are processed. So I think you probably need to open a separate issue

Rokory, 2022-01-08:
Maybe for someone reading this thread, it is of help, but I fixed my problem:

If you use a USB-based BT stick, in Control Panel, in Power Saving, in Advanced settings, under USB settings, you have to disable selective USB power saving. I do not have the exact wording for English versions of Windows, since mine is German, but I think, you can find the option.

ThiagoAnd, 2022-02-09:
@AeliusSaionji , any news from reWASD team? I am with the same issue.

AeliusSaionji, 2023-07-18:
@ThiagoAnd reWASD actually went and implemented their own driver. Nothing fancy now, more features planned (trackball emulation!)

ThiagoAnd, 2023-08-11:
@AeliusSaionji , so now it's working fine?