imbushuo/mac-precision-touchpad #427: Stuttering audio on headset when using with touchpad

cnadolny2s, 2021-08-06:
When using together with a bluetooth headset, the audio has a constant stuttering. When turning off the apple touchpad, audio becomes fine again. There seem to be some interferences between the two BT devices. My BT dongle uses BT 4.0. I will try to upgrade it to BT 5.x

Does anyone have a similar issue when using bluetooth audio in parallel?

Trevor-Z, 2021-08-12:
I have the same issue.

Tested this on my PC desktop, with JBL Tune 215BT earbuds and Orico bta-508 USB dongle, both BT 5.0.

When I play a video on Youtube or VLC and use the MT2 via the BT driver, after a while the audio will begin to stutter. I turn the trackpad off with the physical button, and soon the audio returns to normal. Turn it on again, the stutter comes back.

imbushuo, 2021-08-15:
This trackpad uses classical Bluetooth and therefore it's a bit expected to have that. I will see what I can do here.

cnadolny2s, 2021-08-20:
My issue somehow disappeared and I can not reproduce it anymore. I have paired the MT2 again and I've also plugged my BT dongle to another USB port. Later I've also switched over to a new BT 5.0 dongle. Not sure what really fixed it in my case.

trankillity, 2022-02-06:
Just came to report the same issue. I am using a BT4 dongle that is in the front USB port giving direct LOS to all BT devices with a distance of 30-60cm. It only has 3 devices paired to it:

- Keyboard (nice!nano based custom keyboard)
- Headset (Sony WH-1000XM3 - likely a few endpoints because it also supports telephony)
- Magic Trackpad 2

I would rather not have to leave the Trackpad plugged in, but it's currently not usable over BT due to this. I might try grabbing another BT5 dongle and seeing if that fixes my issue as @cnadolny2s claims it may have.

idan, 2022-02-08:
Does the magic trackpad 3 solve these issues?

trankillity, 2022-02-08:
> 

No. I have one.