imbushuo/mac-precision-touchpad #428: Disabling Palm Rejection to allow usage of disabled persons.

AlseinX, 2021-08-07:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Yes, it is quite difficult for my girlfriend who was born with no hands to use the trackpad with her stump (that is rejected as a 'palm').

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
Provide with an option to disable the palm rejection and gesture feature and turn all movements into simple cursor moving operations.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
No other idea.

**Additional context**
Add any other context or screenshots about the feature request here.
No.

MagicUtilities, 2021-08-10:
This driver only provides the actual touches to Windows, no further processing is done. Palm rejection is then part of Windows touchpad processing, The internal tweaks are not really documented as far as we know. Nothing this driver can really do here.

Consider to try the [Magic Utilities](https://magicutilities.net/magic-trackpad/features), they will accept any touch to move the mouse cursor.

Hope this helps.

AlseinX, 2021-08-13:
> This driver only provides the actual touches to Windows, no further processing is done. Palm rejection is then part of Windows touchpad processing, The internal tweaks are not really documented as [...]

Thank you very much!

BTW, is there any open-source alternative?