imbushuo/mac-precision-touchpad #429: Question - Are there settings available in Windows 10 for Apple Trackpad 2

stephen147, 2021-08-09:
I saw this article here: https://techzoneonline.com/how-to-use-apple-magic-trackpad-2-on-windows-10/ that shows the settings below.

Do I get these settings available when the device is connected?

[image]

lloydo3000, 2021-08-09:
Yes

stephen147, 2021-08-09:
> Yes

Thank you. That's a great help. I was looking at this https://magicutilities.net/magic-trackpad/features but it's 10eur a year for a driver! I don't know if there are more settings available with that but I'll try this out when I go buy it.

These are the settings for that.
[image]

imbushuo, 2021-08-15:
As other answered, yes (all PTP trackpad will have that)

closed 2021-08-15 by imbushuo

stephen147, 2021-08-17:
Thanks for letting me know. So most of the options that are available with that magic utility app will be in this driver?

lloydo3000, 2021-08-17:
Everything from the Windows settings page for Precision touchpads will be there. Let us know if you need a screenshot

stephen147, 2021-08-17:
Yes, that would be great of you. Thanks.

lloydo3000, 2021-08-17:
[image]

stephen147, 2021-08-17:
Thank you. It seems to have all the important features that the app at 10 eur a year has.