imbushuo/mac-precision-touchpad #432: 3 finger drag

p2rkw, 2021-09-02:
**Pre-report checks**
- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]
Hi, thanks for sharing this cool project. 
I came with a question: is it possible to add more config options to this driver or are we limited by windows settings? I'd like to add 3 finger dragging to trigger button down on start and and button up when drag finish. This way I could record some gestures and map them to actions with external soft like strokesplus.

On linux I achieved this by modifying driver code, here's my repo: https://github.com/p2rkw/xf86-input-mtrack

Is it even possible to extend this driver like that, or is it out of scope of this driver. Where should I start to get this done?

**Describe the solution you'd like**
A clear and concise description of what you want to happen.
3 fingers on the pad -> trigger button down
3 fingers of the pad -> trigger button up

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
None so far, Im new to windows.

**Additional context**
Add any other context or screenshots about the feature request here.

On linux I use my modified driver which trigger whatever button I configure, then use easystroke to map gesture to actions/keyboard strokes.

adventures, 2021-09-18:
Both of these ideas are excellent.

leoclarkribeiro, 2021-11-17:
I live for 3 fingers drag. Will try this. Thanks so much!

Mortenest, 2022-01-04:
Also seems to be the only thing I need extra for this awesome project. 🙌

maratori, 2022-08-05:
Totally agree, this is the only missing peace.

leoclarkribeiro, 2022-08-06:
Thanks for the support guys.
It seems the problem is Windows that does NOT have the option.
I am about to sell my PC and go back to Mac.
Cheers

MrWe, 2023-03-23:
@p2rkw Did you find a solution to this? 3-finger drag would make this perfect.

leoclarkribeiro, 2023-03-26:
Best of luck guys.
I am back on a Mac...
I can't live without 3 fingers dragging anymore.

estenger, 2024-08-28:
3 finger drag on PC is the only missing feature here IMO. Here's a pic of me waiting for this feature add:

[image]

Real talk though. I've never worked with drivers for PC before. Is it possible to do custom configuration of the driver to enable this? Or would this have to be a hard-coded configuration within the driver files?

jirikrepl, 2024-12-02:
I think I am using Bootcamp drivers, 
but this utility might work with these precision drivers.

https://github.com/ClementGre/ThreeFingerDragOnWindows

I believe this app converts three-finger drag to one-finger tap and drag.
For me, the three-finger drag with this app is a little too fast,
but it might be possible to adjust it in the app settings, which I haven’t tried yet.