imbushuo/mac-precision-touchpad #438: Touch, move then push to click jumps the cursor to its previous position.

glcraft, 2021-09-27:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

When I put my finger to move the cursor, move the cursor a bit then click by pressing on my Magic Trackpad, the cursor jumps on its initial position before I touch the trackpad.
If I do the same but with larger movement, it does not jump.

Here is a screen capture :

[image]

I'm using Windows 10, it does on my 2 PC, in both USB and Bluetooth mode, I installed the last update of your driver, it does this since I installed first the driver (I mean that it always did this, but I just decided to make an issue).

It may be a threshold to click on the right place, but it uses to be very annoying to click on things close to the cursor because of that 😅

tritegeist, 2021-09-27:
I can reproduce this as well on Magic Trackpad 2. From what I can tell, the threshold is actually time-based rather than distance-based. If you click on the trackpad immediately after your finger touches the surface it will jump back to where you originally started. I'd say this happens within 1/4 of first touching the trackpad maybe even less (1/10th maybe? It's very brief)

glcraft, 2021-09-27:
Hmm, you may be right. It seems to happen within half a second.

lloydo3000, 2021-09-28:
I can reproduce this as well

trogper, 2021-12-28:
@Amir-19 @watercat12 dont use "same here" on github. To reduce clutter, use +1 / like buttons.

I have seen this behaviour on a few other laptops, so it might also be (partly) Windows issue

Colbey, 2022-04-14:
one "work around" I have found is to actually lift your finger up before clicking. does not work all the time but a good portion. 
another "work around" I use is just click somewhere else on the touchpad. This usually works 100% of the time.