imbushuo/mac-precision-touchpad #450: [Windows 11][Bluetooth Connection] Trackpad 2 suddenly stop responding until turn off and turn on again

chinhln1902, 2021-11-02:
First thing first, thank you very much for your effort to make the Trackpad 2 works on Windows OS. I appreciate your work!

I have found #423  issue but seem like the person in that issue used wired connection and there is no solution in that issue yet so I report separately.

**Describe the bug**
I am using Magic Trackpad 2 on Windows 11 using Bluetooth connection. The problem is sometime the trackpad stop responding as I couldn't move the cursor (tried two, three, four fingers moves but didn't work). So I turned the trackpad off and turned it back on and it works fine.

I haven't tried to use cable so I don't know this could happen on wired connection or not since I prefer the wireless connection.

**To Reproduce**
There is no exact steps to reproduce the bug. It just sometime happens when I left it idle for long time because I am using another wireless mouse beside the trackpad as well.

So I assume the steps to reproduce it is:
1. Connect the trackpad 2 to PC using Bluetooth connection.
2. Leave it idle for hours then sometime move the cursor to check. If the cursor moves then it is still working. Otherwise the bug shows up.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.
My OS is Windows 11 and I'm using trackpad via Bluetooth connection.

Thank you,
Chinh Le

lloydo3000, 2021-11-02:
This happens to me on Windows 10 but all I have to do is click (not tap) and it wakes up

chinhln1902, 2021-11-02:
> This happens to me on Windows 10 but all I have to do is click (not tap) and it wakes up

I’m not sure because of I’m using Windows 11 or not. But I tried every way: click, tap, two, three, four fingers tap and click, etc but it didn’t move

egorgam, 2021-11-13:
I have same behavior in Win11 - Magic Trackpad 2 is not responding after reboot. And It fixes after removing trackpad from Bluetooth devices, and add again

mohammadjavad75, 2022-03-30:
I have the same issue

ScottKin, 2024-01-15:
Mine is worse...

Changing batteries and my Trackpad v1 refuses to connect whatsoever - even thinking this is it's first connection, needing passcode, connect / not connected / connecte / not connected.