imbushuo/mac-precision-touchpad #455: Install on Bluetooth. USB works fine

JolleSc, 2021-12-08:
Hello,

For bluetooth I can not choose one of your drivers in windows which will work.
Only the Apple Multi-Touch Pro driver is accepted when the trackpad is connected by bluetooth.

Which driver I should use for the bluetooth connection

jankolkmeier, 2021-12-27:
I had the same issue, would pick this driver when plugging over USB, but would use some other default driver when pairing through bluetooth. My solution was to manually set the drivers to the ```AmtPtpDevice.inf``` one using this procedure:

1. (after it's already paired) go to to the Windows "Devices and Printers" windows, find your trackpad, right-click properties, and go to the hardware tab.
2. For each device listed in the Hardware tab, go to that devices properties, go to the "Driver" tab, and update. Don't use the automatic search, but use the "Browse ..." option. 
  - Point it to wherever you've installed these drivers (you can just point it to the unzipped copy of this driver). 
  - If you can't "Update Driver", make sure to elevate permissions on the "General" tab first
  - It may state that it doesn't have a more up-to date driver for some of the devices in the list. I've just ignored it and continued for the other devices

For me, no restart or re-pair was needed, it just worked once I was done manualy updating the drivers with this method.

JolleSc, 2021-12-28:
> EXTERNAL: This email originated from outside of the organization

Thank you.
I was looking for the wrong driver.  I did not see that I had to choose ‘Bluetooth HID-Device”

It works now perfect!  Thank you so much for this!

Met vriendelijke groet,
Kind Regards,

Jolle Schrale
Asset Manager | Faber Audiovisuals
NEP Live Events | The Netherlands
O:+31 883 223 700 | M:+31 629 537 068
***@***.***
www.faber-av.com<http://www.faber-av.com>

Superior Service & Lasting Relationships
Integrity | One Team | Innovative | Passion

CONFIDENTIALITY NOTICE & DISCLAIMER: This email may contain confidential, proprietary, or privileged information. If you are not the intended recipient, please contact the sender by reply email and destroy all copies of the original message.

ThiagoAnd, 2022-02-08:
Hi! Im new with trackpad, I have the same issue, installed the AmtPtpDevice.inf and works on USB, but when I try to use bluetooth, no response from the magic trackpad. I have to install another driver? Which one? Can you send me the link?

ThiagoAnd, 2022-02-09:
@jankolkmeier , after installing the AmtPtpDevice file, these drivers appears. I can use USB fine, but Im unable to use bluetooth

[image]