imbushuo/mac-precision-touchpad #457: Doesn't respond to input on Magic Trackpad 2nd gen with Bluetooth connection

m13253, 2021-12-11:
**Pre-report checks**

- [X] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
The driver doesn't respond to input on Magic Trackpad 2nd gen, connected with Bluetooth connection.
Device manager shows STATUS_DEVICE_POWER_FAILURE.

**To Reproduce**
1. I paired the trackpad before installing the driver
2. Install the driver by right clicking on the INF file
3. Delete the pair and re-pair the trackpad
4. Wait for some minutes, Bluetooth settings says connected
5. Touch the trackpad, but the mouse cursor on my computer doesn't move
6. Restarting both devices can't fix the problem

**Expected behavior**
The mouse cursor on my computer should move.

**Traces**
[image]

**Environment**
Windows 11 Version 21H2 Build 22000.348
Magic Trackpad 2nd gen (A1535)
Connected through Bluetooth

imbushuo, 2021-12-11:
Confirmed to be an Windows 11 specific issue and starts investigating.

ruipires, 2021-12-21:
I seem to have the exact same issue (same symptoms), and I'm on windows 10 (Version 1909 build 18363.1977).

Zerov-ua, 2021-12-23:
same. w11

malcolmturnbull, 2021-12-29:
I am also experiencing this issue on Windows 11. 

`Device HID\{00001124-0000-1000-8000-00805f9b34fb}_VID&0001004c_PID&0265&Col01\9&11cd7162&5&0000 had a problem starting.

Driver Name: oem57.inf
Class Guid: {745a17a0-74d3-11d0-b6fe-00a0c90f57da}
Service: mshidkmdf
Lower Filters: AmtPtpHidFilter
Upper Filters: 
Problem: 0xA
Problem Status: 0xC000009E
`

ThiagoAnd, 2022-02-09:
same here w10. I manage to get bluetooth to work, removing rewasd. I will try more tests. Any workaround?

puretruth, 2022-10-31:
Same problem on Windows 10
System massage same as @malcolmturnbull pasted.

tanisou, 2023-05-08:
Does someone solve this problem?
I'm facing same problem on Windows 10 with Magic Trackpad 2.

Fire-, 2023-08-30:
Same issue here - no real way to dig into it either.