imbushuo/mac-precision-touchpad #458: Erroneous right clicks on 2012 MacBook Pro caused by inability to turn off 2-finger right click

Browningate, 2021-12-14:
I've combed through the bug reports, but didn't see anything about this exactly. It appears that the 2-finger right click function cannot be turned off. As a result, about 30% of all left clicks are registered as right clicks.

**To Reproduce**
After turning off all other gestures, clicking the pad results in a right-click being registered about 30% of the time. If I intentionally place 1 finger on the touchpad, without moving, and then click it with another fingertip, a right click is registered 100% of the time. It does not appear that there is any way to replicate this.

**Expected behavior**
No right-click unless I single click toward the lower-right corner of the pad

**Traces**
If it would be helpful for me to collect these in order to help diagnose the bug, feel free to respond to the report to let me know.

**Environment**
Windows 10 build 17763 running on MacBook Pro 2012 with Retina ("MacBookPro 10,100"). Using the built-in trackpad. This behavior is completely absent in Apple's stock driver, (unless I enable that "feature" specifically) so I have to assume that we aren't dealing with a hardware or behavioral problem.

qiukeren, 2022-03-26:
seems like #256

Browningate, 2022-03-26:
> seems like #256

Not really.

MoAnsir, 2022-12-02:
I Have a similar problem. But for me, it happens when I scroll up and down a page. I will be scrolling the page and about 40% of the time it will pick up a right-click and open the menu.