imbushuo/mac-precision-touchpad #461: The cursor has drift and hysteresis when moving when I use trackpad in windows 11

guohui-ma, 2022-01-04:
just as title describe.

mikebeaton, 2022-03-10:
I wonder if this is the same issue as I am seeing, since hysteresis could be about many aspects of the movement.

The specific issue I see is that after doing a 2 finger scroll, when I remove the second finger and try to make some next action with one finger, then for a noticeable time (maybe 0.5s?) any action with 1 finger is still captured as a scroll (i.e. as if 2 fingers were still on the trackpad). So for me, the hysteresis is in the timing of detection of 2 fingers on the trackpad, and it would be good to have some way to control the sensitivity of this.

jasonm23, 2026-01-19:
If installing via `choco` an error is thrown that Windows 11 is not supported, I would guess this is skipped if installing manually.