imbushuo/mac-precision-touchpad #462: Doesn't work after install

Zingo21, 2022-01-17:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

Yes and I can't find the issue that I have experienced 

**Describe the bug**
A clear and concise description of what the bug is.

After I've installed the driver in windows 10, I couldn't do the gestures like you could in MacOS. I've tried the latest release and older versions but it doesn't work. I've also tried to install the driver again with no success. 

**To Reproduce**
Steps to reproduce the behavior:

Download whatever of the two ZIP-files provided. Unzip them and install the driver.

**Expected behavior**
A clear and concise description of what you expected to happen.

I was expecting that after install that the Mac gestures would work with the activity field and the usual stuff that you could do in MacOS

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.
MacBook Pro 2017 13 inch. Apple SPI Trackpad. Windows 10.

martinsstuff, 2022-01-22:
You can configure certain touch gestures in the Settings app: https://support.microsoft.com/en-us/windows/touch-gestures-for-windows-a9d28305-4818-a5df-4e2b-e5590f850741

If you can't find the "Touchpad" option, the driver may not have been installed correctly.

Zingo21, 2022-01-24:
Ok! Thanks for the link. I notice that when I install it that it comes a notification that says that the driver is installed. But when I look at the drivers installed, then I don't see the driver that I just installed. So it's like that it was installed correctly but not.

closed 2022-03-15 by Zingo21