imbushuo/mac-precision-touchpad #463: Ability to toggle touchpad delay after typing

marc2323, 2022-01-27:
Touchpad is dead for 1-2s after using the keyboard. Makes it impossible to make quick cursor moves without waiting for touchpad to activate again. 

Proposing ability to toggle delay after typing (0 seconds).

ThiagoAnd, 2022-02-08:
Hi Marc, I just installed the drivers here and had the same problem. The only way I've found to resolve this is to set the touchpad sensitivity to the highest. So you can type and move on the touchpad at the same time.

RobertPotter, 2022-05-09:
Hey Marc, I managed to resolve this without having to set the sensitivity to its highest. Changed a registry value as detailed here:
https://www.howto-connect.com/fix-synaptics-touchpad-delay-after-typing-windows-10/

- HKEY_CURRENT_USER > Software > Microsoft > Windows > CurrentVersion > PrecisionTouchPad
- Switch to the right pane, locate and double-click the DWORD AAPThreshold.
- In the Value data field, insert 0 and click OK
- Restart Windows

ThiagoAnd, 2022-06-13:
> Hey Marc, I managed to resolve this without having to set the sensitivity to its highest. Changed a registry value as detailed here: [...]

Great! I will try too. Do you experienced sometimes 2 fingers command doing 3 fingers command instead? Sometimes when I try to scroll down the page with 2 fingers, the trackpad somehow understand that I doing the 3 fingers scroll down, click,.... I believe its 30% of the time.