imbushuo/mac-precision-touchpad #467: I can´t use magic trackpad 2 with bluetooth on Windows 10

ThiagoAnd, 2022-02-09:
Downloaded the driver .zip, executed the AmtPtpDevice file, works ok with USB, but not with bluetooth.
Paired the trackpad, but unable to use it. It seems there is a issue for windows 11 on issues list, but mine is Windows 10:

[image]
[image]
[image]

ThiagoAnd, 2022-02-09:
Just found the bug for me. Im using rewasd. After deleted rewasd, the Apple Multi-touch Trackpad HID Filter, work fine, and the bluetooth start to work as well. 

So, for me, without rewasd installed:
just download the driver, connect the trackpad with a cheap bluetooth adapter and it will work.

@imbushuo , just asking, do you know if this bug its easy to be fixed? I am contacting also the rewasd team, to see if there is a fix that could be done on their side.

sjain882, 2025-07-20:
Hi, I'm having a problem where I can't even pair the trackpad to any Windows machine. It just unpairs immediately.

This is completely independent of any issues with custom drivers - it won't pair even on a fresh machine, just to test it with basic support (no precision, left click only). I don't even get a chance to install any drivers.

It pairs fine on macOS.

Full details here: https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/20#issuecomment-3094498046

Did anyone else have this issue or can advise what they did? I'm completely stumped. Thank you