imbushuo/mac-precision-touchpad #469: Userspace implementation of three-finger-drag via WM_INPUT and mouse_event

xeonmc, 2022-02-10:
## Issue
Windows Precision Touchpad implementation lacks the logic of three-finger drag, and since this driver project is merely a translation layer to connect to the Precision-Touchpad it is impossible to implement it driversie.

## Solution
Have a userspace program that monitors WM_INPUT messages in the background and synthesize mouse_event calls. The program should also allow one-click disabling of all Precision-Driver three-finger gestures as not to cause interference.

## Implementation Sketch
1. Userspace program configures Precision Driver options to "Do Nothing" for all three-finger gestures
2. Userspace program calls RegisterRawInputDevices with the Generic Desktop Digitizer usagepage, and with RIDEV_INPUTSINK + RIDEV_DEVNOTIFY flags
3. Upon receiving WM_INPUT_DEVICE_CHANGE arrival notices, userspace program calls GetRawInputDeviceInfo for each handle received to identify the device that we wish to monitor, as well as the preparsed data so that the program can correctly interpret all incoming WM_INPUT messages.
4. For each incoming WM_INPUT message:
    1. Call GetRawInputData with flag RID_HEADER to check that the handle belongs to the device we wish to monitor, discard if not.
    2. Call GetRawInputData with flag RID_INPUT for messages from the monitored device
    3. Use the preparsed data to interpret the touch state, evaluate whether user is interacting with the three-finger-drag gesture
    4. synthesize the corresponding motion and action (accounting for user sensitivity preferences) using mouse_event, which broadcasts mouse motion rather than sending them to a specific app as SendInput would. NOTE: mouse_event cannot send to higher-privilege processes, so in order to interact with elevated windows the Userspace program must be launched with the same or higher privileges.

EternalHXP, 2022-06-21:
cool!!!

ClementGre, 2023-09-05:
Hi, I developped an app that does exactly this, enabling three fingers drag on any Windows Precision compatible touchpad:
https://github.com/ClementGre/ThreeFingersDragOnWindows.