imbushuo/mac-precision-touchpad #472: Work in USB no un Bluetooth

mrStevenx3, 2022-02-15:
Hello,

I installed the "mac precision touchpad" it works well in USB but does not work in Bluetooth.

When my touchpad connects via Bluetooth, the "touchpad" option disappears in Windows options.

I specify that when I go to the device manager, I do not see "Apple Multi-touch Auxiliary Services" and "Apple Multi-touch Trackpad HID filter" I only have "Apple USB Precision Touchpad Device (User-mode)"

I'm on Windows 11.

Thank you for your help and sorry for my bad English, I'm French ^^

zhyu, 2022-02-28:
Same here. I'm on Windows 11 as well.

tnaseem, 2022-03-18:
Same issue for me. I was hoping to stop using Magic Utilities, ah well.

CaptainHugg, 2022-04-10:
Same for me on Windows 10

Piero24, 2022-05-11:
@mrStevenx3 Have you solve the problem?

nexus, 2022-05-11:
Same for me on Windows 11

Piero24, 2022-05-11:
@nexus I have made several attempts but the problem remains... I hope someone has solved it

zhyu, 2022-07-16:
@mcpguide it works! Thanks a lot!

mcpguide, 2022-07-19:
> @mrStevenx3 @zhyu Have you solve the problem?

@Piero24, Could you try the way I posted, and let me know?

Piero24, 2022-07-20:
Hi @mcpguide , I solved it with a similar [procedure ](https://github.com/imbushuo/mac-precision-touchpad/issues/486#issuecomment-1140337268)

kiroZhang1997, 2022-08-05:
> I had the same problem, and it is working perfectly now after cleaning up the driver. Open "Cleanup Disk" app --> Choose any drive --> Click "Clean up system files" button at the bottom --> choose C: drive --> check "Device driver packages. Then, restart your computer.
> 
> could you let me know if your problem has been solved after this? [image] "

that works for me,thank you!