imbushuo/mac-precision-touchpad #477: (Firmware update induced?) Long bluetooth connection times, doesn't affect "official" drivers or MagicUtilities. (Detailed Report)

martinsstuff, 2022-04-02:
**Background**
I was troubleshooting a random Bluetooth disconnection issue*, during this I ended up pairing my Magic Trackpad 2 (Grey/Black) to my iPad and leaving it for an hour or so, afterwards (I am not 100% sure but the timeframe fits) I started having issues connecting it at all to my PC while in Windows. Throughout all of this I had no issues (besides the now worked-around disconnect bug) in Linux, connecting via USB also works as expected.

*I believe the issue was due to having my monitors USB cable plugged into a certain USB 3.0 port/controller, instead of the previous USB 2.0 port, a certified AMD™ B550™ moment.

**The bug presents as follows:**

USB Connectivity is not affected and behaves as expected.

Bluetooth with no drivers: Pairs successfully but no input. (I might be misremembering but I *think* it used to work? Just without right click or scroll, as does USB with no drivers.)

I took multiple measurements to get an average connection time (or a range), measurement starts from first click (to initiate connection) until cursor begins moving:

* Connection times with official drivers: ~6 seconds
* Connection times with this driver: ~19-20 seconds (occasionally as fast as 15 seconds)
* Connection time with MagicUtilities: Basically instant. (Unfortunately they don't use Windows Precision 😞)
* Connection time on iPad/MacBook Pro: Basically instant.
* Connection time on Linux with built-in "hid-magicmouse" kernel driver: *Slower* than MagicUtilities (on Windows) but still fast enough that I don't consider it a "delay".

**Attempted Troubleshooting**
* Full reinstall of Windows operating system (multiple times, actually).
* Installing Windows on my MacBook Pro via Bootcamp (same issue)
* [Factory (Firmware) Reset](https://apple.stackexchange.com/a/435397) of Magic Trackpad 2 via Mac (option was removed in Monterey, had to install Big Sur on a different partition)
* The same(?) type of reset via MagicUtilities ("Full factory reset")
* Updating/Downgrading Motherboard/BIOS Firmware.
* Repositioning the Wireless/Bluetooth antenna.
* LOTS of reconnects, driver un/reinstalling (including Wireless/Bluetooth) and a bunch of other stuff.

**Expected behavior**
I don't remember it ever taking so long to connect, I have been using this driver for quite a while now and I'm pretty sure it never took *that* long. 20 seconds might not be "long" but it's annoying to have to wait when, for instance, starting the PC or just generally having to reconnect the Trackpad.

**Environment**
OS: Windows 11 (Windows 10 LTSC was also tested, and presented the same issue)
Platform: AMD B550 (Gigabyte B550 Vision D/Ryzen 5900x)
Wireless/Bluetooth: Intel AX200 NGW

Magic Trackpad 2 "Space Grey"
Info from iPad:
* Firmware Version "0860.0398.170"
* Hardware Version "0001"

Info from MacBook:
* Firmware Version: "1.7.0"
Note: I didn't check the firmware before the bug started.

**Random stuff that I noticed:**
* This shows up shortly after initiating the connection but goes back to "working normally" a while before the mouse cursor starts moving: [image]
* The "Touchpad" menu appears right after initiating the connection.
* Movement and clicks during the 20 or so seconds are being stored in some manner, as they rapidly "play back" when the device is fully connected.

provalinf, 2022-09-10:
Interesting subject, I have unpacked the last trackpad drivers delivered with bootcamp 6.1.19, the drivers carry the date of 04/07/2022 and it seems that apple is made of apple by forcing the disconnection of the trackpad just after pairing, basically it is no longer usable, maybe it controls that the pc is really a bootcamp by registry key or by bluetooth chip.  
Another hypothesis is that this version of bootcamp comes with a changelog that says it improves the bluetooth connection, maybe these improvements make it no longer compatible with the bluetooth of my intel AX201.
[latest bootcamp 6.1.19.zip](https://github.com/imbushuo/mac-precision-touchpad/files/9540097/latest.bootcamp.6.1.19.zip)

martinsstuff, 2022-10-23:
@provalinf Do the old drivers still work, or was the firmware changed again? I attached the old files; had to zip them for GitHub, the unpackable executable inside is signed by Apple.
[AppleBcUpdate.zip](https://github.com/imbushuo/mac-precision-touchpad/files/9846956/AppleBcUpdate.zip)

martinsstuff, 2022-10-23:
**I totally forgot to mention:** THE BUG FIXED "ITSELF".

I connected my Magic Trackpad to my iPad via Bluetooth and left it idle for an hour or so, afterwards the firmware number changed and everything worked as expected again.

WinkelCode, 2023-01-02:
@provalinf Not sure if you still have the issue, but I figured out that with the newer driver, you simply have to make sure it's not installed while pairing, afterwards you can install it and everything seems to work fine.

I have a work-in-progress script to automate this: https://gist.github.com/WinkelCode/cae44ddce205b5363c60f15b2748ad1e

Right now it just uninstalls the driver using pnputil.

provalinf, 2023-01-02:
@WinkelCode Sorry I didn't reply, I forgot about this thread!  
Can you show the version of your current firmware?  
I'm going to try again and see if my trackpad updates with the trackpad connected in USB on my macOS VM as I don't have a mac on hand.  
But to answer your question of October 23, during my test I had removed the old bootcamp driver (which worked perfectly) via pnputil, reconnected the trackpad without driver, there it worked in the classic way in limited mode, then by installing the new bootcamp driver, the trackpad disconnects. Even when unpairing and re-pairing, the trackpad connects and then in the second disconnects.

WinkelCode, 2023-01-03:
@provalinf I made a new version of the script: https://gist.github.com/WinkelCode/faa1d6000a4ec379e039110cbf97b8de You have to put it into the extracted "AppleBcUpdate" folder. (You don't have to use it, but it made the process a lot faster for me, and also *should* prevent accidentally removing the wrong drivers, still, use at your own risk!)

Here is what worked for me:
(BT = Bluetooth settings in Windows)

1. Remove Trackpad from BT if it is still there
2. Disable BT
3. Remove Trackpad drivers
4. Enable BT
5. Pair Trackpad (it works, but only basic functionality)
6. Disable BT
7. Install Driver
8. Enable BT
9. It works for a moment, then it stops and I get a notification to restart the PC
10. Turn trackpad off/on
11. Works as expected now

My iPad reports Firmware ver. `0860.0398.0192`
MacBook Pro on Ventura reports: `1.9.2`

Observation: The firmware versions from my first post make me suspect that 
`0860.0398` is static and `0192` equals `1.9.2` on iPad/macOS respectively.