imbushuo/mac-precision-touchpad #478: Support Apple Magic Trackpad 1 (MC380LL/A)

Masiha, 2022-04-04:
I wonder would you have a plan to support the first generation of the magic trackpad?

carlitorendon, 2022-05-03:
Totally give my vote for this. I have two of these. @imbushuo I'm happy to ship one 🙂

fschuh, 2022-05-13:
I was disappointed to find out this only works with Magic Trackpad 2 or newer...
Hopefully Magic Trackpad 1 gets support soon as well!

HubKing, 2022-08-27:
Any comment from the developer(s)? Is it impossible or is there any hope? I mean, if it is very difficult so that there is very little chance that it would ever be supported, it would good for us to know the fact so that we would not hope.

HubKing, 2022-08-27:
Seems like this is a duplicate: https://github.com/imbushuo/mac-precision-touchpad/issues/358

The developer even purchased TP1 two years ago, so I guess it was really difficult for him and gave it up.

dmitry-entin, 2026-10-29:
Draft PR with Magic Trackpad 1 (Bluetooth) support: #579

Heads-up: it was written with an AI assistant and tested on one computer only (Windows 11 x64, MT1 over Bluetooth). Cursor, tap, scroll, pinch to zoom and 3/4-finger gestures work there. It is not officially signed, so trying it needs a self-built driver and test signing enabled (which requires Secure Boot off).

If you try it, please report your results in the PR, especially on other hardware.