imbushuo/mac-precision-touchpad #480: Magic Trackpad 3

THALILNINJA, 2022-04-17:
Hello, 
is this working on Magic Trackpad 3 ?
Are you considering releasing an updated version for it ?

oddko, 2022-04-18:
I have one and it works just fine with all 3&4 finger gestures.

rdhar, 2022-04-19:
This is the first I'm hearing of Trackpad version 3: would you mind sharing a link to product page to review/compare specs?

oddko, 2022-04-20:
Apple still calls it the "Magic Trackpad" but you can notice it with the new packaging (braided lightning cable, etc.) and also the return of the black edition alongside the recently released Mac Studio.

spex66, 2023-04-25:
> I have one and it works just fine with all 3&4 finger gestures.

@oddko what do you mean exactly?
a) do you use the same driver as for Trackpad 2 and it works for Trackpad 3 too?
b) or you have another driver, which works?

If b) => where is it available for installation?

Thanks for your reply!

oddko, 2023-04-27:
Hi, it worked fine with the driver as-is, installed it with Chocolatey.