imbushuo/mac-precision-touchpad #482: Haptic feedback magic trackpad 2 on windows 10/11 - non mac pc

whitefoxsk, 2022-05-04:
Bought used magic trackpad 2 , used your driver on non mac computer. 
Everything works except haptic feedback. 
When you click you should feel the trackpad response that you clicked or sound, am i right ?
Now i don´t know if it should work on regular pc or not.

MagicUtilities, 2022-05-04:
The haptic feedback should always work, even not connected to any computer. When the Magic Trackpad is turned on you should be able to click. 

Most likely your internal battery is dead, then the haptic feedback doesn't work any more, even the Magic Trackpad is powered by an USB connection.

Hope this helps.

whitefoxsk, 2022-05-04:
> The haptic feedback should always work, even not connected to any computer. When the Magic Trackpad is turned on you should be able to click.
>
> Most likely your internal battery is dead, then the [...]

You are probably right, but can it still be the battery when i can use the trackpad wirelessly via bluetooth?.
In Magic trackpad utilities 3.1.2.6 its says No battery state avaible [64] 
[image]

MagicUtilities, 2022-05-04:
If it works with Bluetooth then obviously the battery seems to work in general. Battery state 64 is unknown to us, only Apple knows what that means. There might be an issue with the battery and/or the haptic feedback hammer.

whitefoxsk, 2022-05-06:
You was right, returned used trackpad 2 , bought brand new magic trackpad 3 and it clicks with haptic feedback on my windows pc.