imbushuo/mac-precision-touchpad #488: Sensitivity bug sometimes

ThiagoAnd, 2022-06-04:
when I'm using the magic trackpad 2, sometimes I'm scrolling the page or doing another movement with 2 fingers, and it executes another command, as if it were with 3 fingers. For example, 2 fingers down, scrolls down and 3 fingers down minimizes windows. So when I scroll down, sometimes it's minimizing windows.

swithdrawn, 2022-06-08:
I have this issue often and ended up disabling three-fingered gestures entirely.  I run a hackintosh on the same PC and on macOS the trackpad works as expected.

ThiagoAnd, 2022-06-13:
> I have this issue often and ended up disabling three-fingered gestures entirely. I run a hackintosh on the same PC and on macOS the trackpad works as expected.

Understood. I believe I have to disable it too. Did you experienced this issue only with 2 fingers -> 3 fingers or also with 1 finger command -> 2 fingers command?

Andrey-Solovyh, 2022-07-25:
I run into this problem all the time

jjoshrtca, 2022-08-25:
It's because the driver still has sensitivity issues.  I used to use this on my old pc until I had to switch to some other driver because the cursor was far too jittery and erratic for Excel/ERP-spreadsheet work (can't remember the name of that unfortunately -- the dev posted it here but got blasted for "advertising" so he deleted his replies/posts -- but it's a full on program that emulates the mouse instead of using Windows crap "precision" drivers).  It even has more gestures...
I recently switched to a new pc and decided to try this again and all the same issues are still there (this that you described, jittery/erratic cursor, cursor still moving when finger's hovering too close to trackpad (but not touching it -- thought it was static but does it with dry and moisturized hands, shoes/socks on and off, etc.).  It's subscription based if that helps anyone looking for a more stable driver.

lsaadeh, 2022-08-25:
I also came here to report this issue.. it seems like palm rejection is something that's also needed. I think most of the issues come from fingers being too close to the trackpad and being recognised.

bradharms, 2022-09-04:
[This post](https://github.com/imbushuo/mac-precision-touchpad/issues/405#issuecomment-932736153) resolved the issue for me.