imbushuo/mac-precision-touchpad #490: Full Palm on Magic Trackpad 2 Disables Trackpad Usage

thehafenator, 2022-06-16:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.

thehafenator, 2022-06-16:
On Windows 10, over SPI, the Magic Trackpad 2 functionality freezes when a full palm is placed on the trackpad, and the haptic button is clicked. I'm on a Lenovo C740 Laptop. 

Then, the computer or the trackpad need to be restarted. I've occasionally rested my palm on the trackpad, and this happens. Not too annoying, though I thought I would place it here.

provalinf, 2022-08-31:
Indeed, I noticed the same problem