imbushuo/mac-precision-touchpad #497: Unable to install driver via .inf file in Windows 11

jfalava, 2022-07-29:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
I've tried to install the driver via the .inf file. When right-clicked and installed, the window that should apperar with a confirmation of a successful install does not appear.
When using `RUNDLL32.EXE SETUPAPI.DLL,InstallHinfSection DefaultInstall 132 path-to-inf\infname.inf` it just shows an error message with no explanation.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.

Windows 11 with the following patches: KB5013889, KB5015882, KB5016353, KB5015898

jfalava, 2022-07-29:
I've tried the Chocolatey driver install and it works with no issue.

RobertFlatt, 2022-08-26:
For the the trackpad had to be connected for the Windows install to work.

sgzman, 2022-08-31:
> I've tried the Chocolatey driver install and it works with no issue.

it works ,ths

tylerrop, 2022-09-09:
Was able to install release 3979 on my personal PC using the INF file in the zip of the release (downloaded straight from GitHub). On my work PC I encountered the same issue, so I had to install Choclatey and then install the drivers package and it worked right away!

jhubsil, 2022-09-19:
For anyone who's successfully installed on Windows 11, could you provide the steps you took?  I've been trying, with no success.

RobertFlatt, 2022-09-19:
The only thing that was not clear in the instructions is the trackpad must be connected with a physical usb in order for the install to proceed.

As far as I remember there was no confirmation dialog, just go to windows settings to confirm installation.

jhubsil, 2022-09-20:
I'm trying to connect a 2009 Trackpad 1 which is Bluetooth only (no USB to connect with) - do these drivers work for that device?

jfalava, 2022-10-24:
Seemed to be a bad Windows 11 installation, as I have tried again in a fresh install and gave me no problems.

closed 2022-10-24 by jfalava