imbushuo/mac-precision-touchpad #498: 键盘和触控板不能同时使用

CWWP, 2022-08-04:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.

CWWP, 2022-08-04:
触控板和键盘不能同时使用

Alex-Lin5, 2022-08-17:
Please report your devices and environment