imbushuo/mac-precision-touchpad #499: Option to use the magic trackpad as touchscreen, not touchpad. Reason: need to process WM_GESTURE messages on application side

svfa001, 2022-08-24:
**Use Case**
We want to use the magic trackpad as gesture input device for a Windows application which is currently in development.
Gesture processing should happen on application side, e.g. the application should receive "WM_GESTURE" messages and be able to react to the different gestures.
For certain reasons it needs to be the magic trackpad.

**Issue**
Currently, and with being registered as touchpad, Windows uses the magic trackpad as a "Precision Touchpad", and somehow "hijacks" all gesture input coming from that device to be mapped to the Windows predefined actions. E.g. three finger tap opening Windows search, four finger tap opening notification center and so on.
No more gesture messages are sent further up the process chain to other applications - which then in turn don't have any opportunity to process those gestures individually.
In contrast to above behaviour, devices which are registered as "touchscreen" are working as intended. With those touchscreen devices "WM_GESTURE" messages are generated and passed over to applications for further processing.

**Suggestion/Idea**
Have a (driver) option to let the magic trackpad be registered as touchscreen, instead of a touchpad. So that gesture messages are generated which then can be processed by other applications.

**Describe alternatives you've considered**
Doesn't necessarily need to be touchscreen. Most important would be to have a driver implementation which allows the gesture messages ("WM_GESTURE") to be passed over to other Windows applications.

tic80, 2022-09-04:
This will be a great enhancement!  I was actually checking for it and while I'm able to register a window for receiving a multi touch input using "RegisterTouchWindow(hWnd, 0))", the windows doesn't receive any events from "WM_TOUCH" while using the magic trackpad.

tic80, 2022-09-05:
Quick update: while it is not possible to get WM_GESTURE message and WM_TOUCH message, it is fairly easy to get HID raw input using WM_INPUT. I'm able to get the position of the 5 fingers in a win32 application with "click" attribute.  Gesture can be easily be deduced from the finger positions.

svfa001, 2022-09-12:
Hi, thanks for the update. Yes, with RegisterRawInputDevices it's working and possible to get WM_INPUT messages. But how did you find out about the raw touch data structure? Honestly I don't have to much insight here and afaik, this also is device/manufacturer specific. Do you have any hint? Did you go byte by byte, figuring out value meanings - or researching the driver code?

ClementGre, 2023-09-05:
Hi, you can look at this example that decodes all of this stuff to get the right position of the five fingers:
https://github.com/emoacht/RawInput.Touchpad