imbushuo/mac-precision-touchpad #500: Magic trackpad 3 lag bluetooth on 2/3 computers

provalinf, 2022-09-09:
Hi guys,
I have a problem with the trackpad v3, the problem does not seem to come from the driver because without any driver installed (which means that the tap to click does not work as well as the scroll and the movement of the cursor is very fast), in short without driver installed, I also encounter this problem.

The problem, only in bluetooth, only on two windows PC (no problem on the 3rd one strangely...) when the finger touches the surface and starts to slide, it can take 1 second or more before the sliding action is taken into account and moves the cursor. The problem also occurs when tap for click, some clicks are ignored, as if there were micro disconnections. I tried unchecking the bluetooth card standby permission, but nothing works, do you have this problem too?

Do you have this problem or solution?

PC 1 : card intel AX201 driver 22.150.0.6 Windows 11 (problem)
PC 2 : card intel AX201 driver 22.130.0.2 Windows 10 (problem)
PC 3 : card intel AX201 driver 22.140.0.4 Windows 11 (good)

Thanks!

provalinf, 2022-09-10:
Thank you for this information,
I had already tested your software, and I just did it again by activating Keep bluetooth connection alive, but there is absolutely no difference, I even set the value to 250ms, no change.
I also added the attribute in the registry and restarted, but I don't feel it makes any difference, I think I'll open the trackpad and try to enlarge the existing antenna by soldering a wire or something.

LoganDark, 2022-12-02:
On my system there is a lag when first touching the trackpad, even when it is connected directly over USB.

It seems like the lag is something like 200ms before the input starts to be registered. So if you put your finger on the trackpad and flick quickly, sometimes it will only register at all or sometimes it will only register the very end of the flick.

Curiously if you flick really *hard* (i.e. click the trackpad at the start) it will always register immediately without the delay.

This may be a Magic Trackpad 2 issue instead of an issue with the driver

kamueone, 2024-12-01:
i have the same issue. After 5 seconds of not touching the trackpad, it needs around 200ms to recognise  movement.

CodingOX, 2024-12-17:
Could you please clarify the following: apart from the potential delay caused by Bluetooth hardware driver issues, under this driver or official driver support, can the USB-C Magic Trackpad (also known as MT3) function properly?

lattecor, 2026-11-13:
I found a way to make my Magic Trackpad feel as responsive as a MacBook's built-in trackpad. I've published a simple utility called [Fast Trackpad](https://github.com/lattecor/fast-trackpad) that dramatically reduces the perceived lag in cursor movement and scrolling, over both USB and Bluetooth.