imbushuo/mac-precision-touchpad #501: Installation in Windows 11

jhubsil, 2022-09-13:
I've been trying to figure out how to install the touchpad in Windows 11, and am struggling to get anywhere.  I've seen references to "use Chocolatey" to install with Windows 11, but I have little to no experience using cmd to do anything.  I think I managed to get Chocolatey installed, then ran the command below, but have no idea what to do from here.  Did it actually install?  Where do I go next to access the touchpad settings?  Is there any way to provide a step-by-step procedure for how to install and use this in Windows 11?

Thanks for any assistance you can provide!

-----------------------------------------------

C:\Windows\system32>choco install mac-precision-touchpad
Chocolatey v1.1.0
Installing the following packages:
mac-precision-touchpad
By installing, you accept licenses for the packages.
Progress: Downloading mac-precision-touchpad 0.2105.3979... 100%

mac-precision-touchpad v0.2105.3979 [Approved]
mac-precision-touchpad package files install completed. Performing other installation steps.
The package mac-precision-touchpad wants to run 'chocolateyInstall.ps1'.
Note: If you don't run this script, the installation will fail.
Note: To confirm automatically next time, use '-y' or consider:
choco feature enable -n allowGlobalConfirmation
Do you want to run the script?([Y]es/[A]ll - yes to all/[N]o/[P]rint): y

Microsoft PnP Utility

Adding driver package:  AmtPtpDevice.inf
Driver package added successfully. (Already exists in the system)
Published Name:         oem183.inf

Total driver packages:  1
Added driver packages:  1
 The install of mac-precision-touchpad was successful.
  Software install location not explicitly set, it could be in package or
  default install location of installer.

Chocolatey installed 1/1 packages.
 See the log for details (C:\ProgramData\chocolatey\logs\chocolatey.log).

pnome, 2022-09-27:
I'm really struggling with this too. Would love to get it working

pix3lize, 2022-10-09:
@pnome @jhubsil Managed to make it works with the video guide from https://www.youtube.com/watch?v=aN3X5h9Vfx0 . So far I tried the Bluetooth connection working fine

f0lie, 2022-10-15:
I installed the application via this method as well. I was a bit confused about the next steps but it turns out it's quite simple. The only next step to do is to pair the touchpad with your computer and then you can start using it. You don't have to reboot the machine at all.

1. Turn on your touchpad by flicking the switch in the top right corner.
2. Go to Settings > Bluetooth & Devices
3. Click on "Add Devices"
4. Click on "Bluetooth" ignoring the other two options
5. Pair with "Magic Touchpad"

Settings can be found by searching for "Touchpad Settings" in the Windows menu. Personally, I am quite pleased with the default settings. Even gesture controls work right out of the box without configuration which shocked me.

Guess4u2, 2022-10-19:
@pix3lize Hi- I am struggling to get it to work with Bluetooth on a Windows 11 AMD Ryzen laptop. The Magic Trackpad 2 works fine wired. With Bluetooth I can pair it but then the trackpad does not react. Funny is too, when in Bluetooth connection: I turn the trackpad off and it still shows as "connected".   I have tried re-installing the latest driver AMD64, and ARM64 and it does not help. 
On the other hand, it works perfectly via Bluetooth with another Windows 10 laptop.... 

Any ideas?

Guess4u2, 2022-10-19:
well... no idea what the problem was,.. in any case, I decided to test the trial version of "Magic Trackpad Utilities" App. Then I might buy the licence for a couple of years. Cheers.

LoganDark, 2022-12-02:
Magic Trackpad 2 works fine wired and over Bluetooth on my Windows 11 22H2 with this driver.

1. Download the latest zip from GitHub Releases.
2. *Extract it*, otherwise the install option won't appear.
3. Right-click the `.inf` file and click "Install".
4. Plug the Magic Trackpad in and then turn it on. It should work perfectly with gestures and everything.
5. Unplug it (leaving it turned on). This will activate Bluetooth.
6. Go to Bluetooth settings and pair with the trackpad as normal.

guillefix, 2022-12-07:
I've tried all these instructions, but gestures and scrolling still don't work when connecting through bluetooth. They only work when connecting via cable. Any idea why?

dimo414, 2023-03-20:
I just tried this on my Windows 11 laptop. Installing `AmtPtpDevice.inf` immediately caused the trackpad (which I'd already paired) to start responding to right clicks, scrolling, and gestures. Seems like it "just works".

@imbushuo maybe you want to mention Windows 11 in the README? I was initially concerned this project was Windows 10-only because 11 isn't mentioned, but it seems like it works.

LoganDark, 2023-03-20:
> @imbushuo maybe you want to mention Windows 11 in the README? I was initially concerned this project was Windows 10-only because 11 isn't mentioned, but it seems like it works.

It also worked for me first-try on 11, both wired and bluetooth, Magic Trackpad 2.

yomoko, 2023-07-20:
I dont see `install `after i right click `AmtPtpDevice.inf`. I am using windows 11. I tried cmd: `RUNDLL32.EXE SETUPAPI.DLL,InstallHinfSection DefaultInstall 132 path-to-inf\infname.inf` doesnt work also. What can i do?

Scotee007, 2024-10-16:
> > @imbushuo也许您想在 README 中提及 Windows 11？我最初担心这个项目仅适用于 Windows 10，因为没有提到 11，但看起来它有效。
> 
> 我在 11 上第一次尝试时也使用它，包括有线和蓝牙，还有 Magic Trackpad 2。

Did you use rewasd? It seems that this application disable magicpad