imbushuo/mac-precision-touchpad #504: Disabled when typing?

GlaciHime, 2022-12-12:
I'm using the Apple Magic Touchpad 2 on Windows 11 and these drivers work fine, except the touchpad disables itself for 1 second whenever I send a keyboard input. For my use case (one-handed gaming) this makes the touchpad unusable. Is this a known issue with the touchpad and/or your drivers, or is there a way to fix this that I'm not seeing myself?

WinkelCode, 2023-01-04:
I believe this is probably a Windows Precision issue, I don't game with it, but it seems like touching the trackpad first, then the keyboard prevents it from locking up. I am using the latest drivers from Apple though.

VrydayVrything, 2023-05-22:
Windows Settings > Bluetooth & devices > Touchpad
Under Taps change Touchpad sensitivity to Most sensitive.
Fixed the issue for me.

CPTNTBS, 2023-07-09:
^ worked for me as well. Don't quote me on this, but I think Windows handles concurrent peripheral inputs in a manner similar to a priority queue, with sensitivity acting as the priority. So with lower sensitivity, other peripheral inputs occur before the touchpad input is executed, hence the freezing of the trackpad.

WinkelCode, 2023-07-09:
> ^ worked for me as well. Don't quote me on this, but I think Windows handles concurrent peripheral inputs in a manner similar to a priority queue, with sensitivity acting as the priority. So with [...]

I'm pretty sure it's just a design choice from Microsoft, on laptops you can accidentally come in contact with the touchpad while typing. Windows also doesn't have any issues with concurrent keyboard and mouse input, which has a much higher polling rate (in my case 1000hz), than the touchpad's ~90hz.