imbushuo/mac-precision-touchpad #505: After clean installation of Windows 10 on laptop (MBP16 Bootcamp), installed the latest build, connected Magic Trackpad 2 with BlueTooth and it recognized as a regual mouse, doesn't function with the windows precision features. How can I solve it?

Rec0cery, 2022-12-24:


JunielKatarn, 2023-01-28:
Did you by any chance install the Apple Bootcamp software before this driver?

chopra2007, 2023-01-28:
No. I just installed the AmtPtpDevice.inf file and installed it as a
Bluetooth device as well. Could the five HID compliant devices (in the HID
sections of device manager) come from me installing the driver? I tried to
remove it from device manager to reinstall, but the word apple isn't
written in my device manager anywhere, and I'm not sure which, if any, hid
compliant device the touchpad is. I tested it on my laptop and it worked,
but on my desktop (where I need it) the TouchPad options aren't even
showing up. I'm using windows 10 pro and I'm grateful for any suggestions
you might have. Thank you!
[image: image.png]

JunielKatarn, 2023-02-10:
Please ping me tomorrow evening.

I'll bring my Magic Trackpad and screenshot the exact drivers that get enabled.

veiledvenus, 2023-03-10:
> Please ping me tomorrow evening.
> 
> I'll bring my Magic Trackpad and screenshot the exact drivers that get enabled.

Did you figure out anything that helped?

JunielKatarn, 2023-03-10:
> > Please ping me tomorrow evening.
> > I'll bring my Magic Trackpad and screenshot the exact drivers that get enabled.
> 
> Did you figure out anything that helped?

I did, but I'd need to be near a Windows machine to give specifics.

Roughly speaking, my problem was that I installed the Bootcamp drivers first, which include Apple's trackpad one.
I found it using Device Manager (sort by driver option) and deleted it.
Then this repo's driver took over and worked.

jdewb, 2023-05-30:
I'm not super savvy with windows drivers, but I tried to remove the bootcamp drivers. I reinstalled your driver. The trackpad connects but it says USB Input Device - Driver error. It works as a pointer, but no scrolling, right click, or gestures.