imbushuo/mac-precision-touchpad #510: 3指在mac下可以移动当前软件位置的，大神，请支持下~

Promise-W, 2023-02-04:
**Pre-report checks**
- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

sntyo, 2023-02-07:
I also want this feature. the gesture that I used frequently
3 fingers hold together to select multilines of the articles for annotations
3 fingers hold together to select a file, drag to the other folder, then drop

Softtanck, 2024-01-19:
I also want this feature too.  Please add it.