imbushuo/mac-precision-touchpad #512: windows11系统中能支持Magic Trackpad的电量显示吗？

leywey0129, 2023-02-11:
**Pre-report checks**
- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
windows11系统中能支持Magic Trackpad的电量显示吗？蓝牙耳机类的可以在快捷按钮中直接显示剩余电量，不知道能否实现妙控板的电量显示功能？谢谢！

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

mnameisjeff, 2023-02-17:
应该是不能显示出来的。

leywey0129, 2023-02-17:
如果能再加上蓝牙休眠时间设置和按压力度调节就完美了。

leywey0129, 2023-02-20:
实际使用中发现电脑蓝牙同时连接trackpad和蓝牙无线耳机的话，trackpad会不定时的有卡顿现象，也就是说有干扰现象，此时关掉其它连接设备仅保留trackpad的连接就没有问题，由于我是放在键盘下方使用的，所以trackpad紧靠着键盘，所以只能用蓝牙方式连接电脑，为此买了一个2.4G的蓝牙耳机解决此问题。

sprite5, 2026-02-09:
> 实际使用中发现电脑蓝牙同时连接trackpad和蓝牙无线耳机的话，trackpad会不定时的有卡顿现象，也就是说有干扰现象，此时关掉其它连接设备仅保留trackpad的连接就没有问题，由于我是放在键盘下方使用的，所以trackpad紧靠着键盘，所以只能用蓝牙方式连接电脑，为此买了一个2.4G的蓝牙耳机解决此问题。

--
usb连接一样有干扰,我电脑不会连蓝牙设备,所以买个蓝牙适配器解决,便是看不到电量,用着用着不动了就有点火