imbushuo/mac-precision-touchpad #513: 3-finger gesture idea

rjsoderlund, 2023-02-13:
I have an idea, since the 3 and 4 finger down are both the same action, show the desktop, is it possible to re-assign the 3-finger down option to minimize the current window?

kleqing, 2023-02-14:
> I have an idea, since the 3 and 4 finger down are both the same action, show the desktop, is it possible to re-assign the 3-finger down option to minimize the current window?

>> You can do it by go to the 'Advanced Feature' in Touchpad settings
>> [image]

rjsoderlund, 2023-02-14:
You’re correct!

I didn’t know about the advanced gesture options!  Thank You!

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F78801337%2F218617697-1a0f9ff3-265f-4265-b1cc-d2478b894154.png&data=05%7C01%7C%7Ce26c989c14b8483184a808db0e2dd2e5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638119362192004776%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=NNgpwKakBNBGDZtkGhU1ZJxh8dFCtNWAvejWpMAI7kE%3D&reserved=0>

mmikeww, 2023-05-10:
> You’re correct! I didn’t know about the advanced gesture options! Thank You! 

close the issue?