imbushuo/mac-precision-touchpad #518: Can't Scroll or Zoom on Bluetooth

Retrobolt, 2023-04-25:
I can get it to work through usb, but not Bluetooth. I can't use it wired with my setup. Can I direct the Bluetooth version to the driver somehow?

backtoaldo, 2023-06-05:
I fixed this problem by installing the driver provided by [apple itself](https://support.apple.com/kb/DL1837?viewlocale=en_US&locale=pt_BR). It worked correctly, maybe this solves it (in this case I use windows 11).

pietrofxq, 2023-06-29:
on windows 10 I cannot get the Touchpad settings to show up. Already tried all drivers available, it connects on bluetooth and with cable but only the cursor works (no scroll and gestures)

pietrofxq, 2023-06-29:
update: I managed to get scroll and zoom working with these drivers http://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe

The trackpad bluetooth does not work anymore though (it connects but disconnects after a few seconds). And there is no way to invert the scroll direction, but it works