imbushuo/mac-precision-touchpad #521: THIS PROJECT IS DEAD

ThiagoAnd, 2023-06-21:
Unfortunately, I believe this project is dead. The last release was in 2021. Does anyone know any application besides this one? I also tried magic utilities.

Trevor-Z, 2023-06-25:
Maybe the author could open the project to the community? There's a fork with some improvements, they could be merged:

https://github.com/1Revenger1/mac-precision-touchpad

ThiagoAnd, 2023-06-27:
@imbushuo please, can you open this project to the community?

SevereMkii, 2023-09-08:
> Unfortunately, I believe this project is dead. The last release was in 2021. Does anyone know any application besides this one? I also tried magic utilities.

While it seems to be currently unmaintained, It's currently working perfectly on both Win10 LTSC and Win11 Pro for me.

Balentay, 2023-09-17:
@SevereMkii the only issue I'm having is audio interference when trying to use my bluetooth headphones while the trackpad is connected via bluetooth

mikhey, 2024-04-04:
If the last release is working properly is there a need for another?  I don't think recent releases/commits is good way to determine if a "project is dead".  If it's working don't fix it, right? 

I've seen similar threads on other projects that were used to basically take over a project with maliciou  intention.  So, I may be a little sensitive to it being just posted here as an issue.  If there is a specific bug like @Balentay mentioned, then that's something to possibly look into. I'm of the opinion, if it works, it doesn't need a release right? If it doesn't, it's open source, pull down a fork, commit some code changes, check it & submit a PR.