imbushuo/mac-precision-touchpad #523: Does this driver still work with Win11?

bonzibudd, 2023-08-30:
I'm considering getting a magic trackpad, this would be the deciding factor for me (Unless there's another method of support for the precision drivers).

kleqing, 2023-08-31:
Yes.

kleqing, 2023-08-31:
This driver still work even on Windows 11 but I did not have a magic trackpad so I'm not sure about it but better try than do nothing!

artistro08, 2023-09-06:
works well for me

thisgitislegit, 2023-09-21:
> I'm considering getting a magic trackpad, this would be the deciding factor for me (Unless there's another method of support for the precision drivers).

Hey there. I currently use the magic touch pad on my Windows 11 desktop and I can confirm that I have it working (with gestures) both wired and Bluetooth. If I had to be nitpicky, I would say that its only ever so slightly more sensitive compared to being used with a mac. If there was a way to adjust this by about 5%, it would be perfect

artistro08, 2023-09-21:
I started using the official drivers from the bootcamp setup and that helped a lot with sensitivity. If you can find them, I'd say use those.

christian-gh-col, 2023-10-09:
> I started using the official drivers from the bootcamp setup and that helped a lot with sensitivity. If you can find them, I'd say use those.

Hello Artistro,

Are you using the official drivers with an Apple machine or a Windows PC?

Thanks,
Christian

christian-gh-col, 2023-10-19:
@artistro08 Thank you, I'll have a look

IlIllIl, 2023-11-10:
Hi, I got `A certificate chain processed, but terminated in a root certificate which is not trusted by the trust provider.` when installing the inf file. What to do next?

rin67630, 2024-09-26:
I could not install on current version of Windows 11 pro 64 22H2, build 22621.4169 using chocolatey:
choco install mac-precision-touchpad

ERROR: Cannot be installed on this version of Windows. Requires Windows 10 x64.
The install of mac-precision-touchpad was NOT successful.
Error while running 'C:\ProgramData\chocolatey\lib\mac-precision-touchpad\tools\chocolateyInstall.ps1'.

???

epetre, 2024-10-03:
> I could not install on current version of Windows 11 pro 64 22H2, build 22621.4169 using an admin terminal: choco install mac-precision-touchpad
> 
> mac-precision-touchpad v0.2105.3979 [Approved] mac-precision-touchpad package files install completed. Performing other installation steps. The package mac-precision-touchpad wants to run 'chocolateyInstall.ps1'. Note: If you don't run this script, the installation will fail. Note: To confirm automatically next time, use '-y' or consider: choco feature enable -n allowGlobalConfirmation Do you want to run the script?([Y]es/[A]ll - yes to all/[N]o/[P]rint): y
> 
> ERROR: Cannot be installed on this version of Windows. Requires Windows 10 x64. The install of mac-precision-touchpad was NOT successful. Error while running 'C:\ProgramData\chocolatey\lib\mac-precision-touchpad\tools\chocolateyInstall.ps1'.
> 
> ???

Same issue here. I don't understand how others managed to do it.

qidi7, 2024-11-13:
> I could not install on current version of Windows 11 pro 64 22H2, build 22621.4169 using an admin terminal: choco install mac-precision-touchpad
>
> mac-precision-touchpad v0.2105.3979 [Approved] [...]

我在win11安装遇到了同样的问题，排查发现是安装前会检查系统版本，通过删除检查的系统版本的代码解决了：

1. find  chocolateyInstall.ps1 in C:\ProgramData\chocolatey\lib-bad\mac-precision-touchpad\0.2105.3979\tools
2.  delete this code:
`# Check OS version - only works on Windows 10
Write-Debug "OS Name: $($env:OS_NAME)"
if ($env:OS_NAME -ne "Windows 11") {
 throw "Cannot be installed on this version of Windows. Requires Windows 10 x64."
}
`
3. run  'chocolateyInstall.ps1'  in powershell
---
1. 在目录C:\ProgramData\chocolatey\lib-bad\mac-precision-touchpad\0.2105.3979\tools 中找到chocolateyInstall.ps1文件
2. 编辑此文件，注掉以下代码并保存（这段代码的用途是检查系统是否为win10）：
`
Write-Debug "OS Name: $($env:OS_NAME)"
if ($env:OS_NAME -ne "Windows 11") {
 throw "Cannot be installed on this version of Windows. Requires Windows 10 x64."
}
`
3. 通过“终端管理员”运行这个文件
> 进入目录： `cd C:\ProgramData\chocolatey\lib-bad\mac-precision-touchpad\0.2105.3979\tools`
> 运行：`chocolateyInstall.ps1`