imbushuo/mac-precision-touchpad #525: windows10 connect Magic Trackpad  successfully when wired，but connect failed when wireless（蓝牙找不到妙控板设备），Does anyone know why?

shiqiongzhen, 2023-09-16:
**Pre-report checks**
- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Donkey-Hao, 2023-10-14:
Do u have other Bluetooth devices in use when u try to connect ur Trackpad? If u have, try turn off all other devices and connect Trackpad first. This is my experience, hope it's usefull.

dynamohum, 2023-11-12:
I've personally never got it work over bluetooth on Windows. Happy enough using it wired but posting here partly to see if you get a fix for it.

shiqiongzhen, 2023-11-12:
小施已收到你的来信，祝你生活愉快~

dynamohum, 2023-11-12:
google translate says "Xiao Shi has received your letter and wishes you a happy life~"

You too matey! :)