imbushuo/mac-precision-touchpad #527: Help with re-installation.

galen-burnett, 2023-11-07:
Please see the details of my issue in this MS Community [post](https://answers.microsoft.com/en-us/windows/forum/windows_11-hardware/magic-trackpad-2-driver-problem/9dd58245-f424-4007-b1cf-fbaa37cc7f57).

dynamohum, 2023-11-12:
Did you fix this in the end? I had a similar issue and just re-installing the drivers brought the magic trackpad 2 back to working order.

galen-burnett, 2023-11-13:
@dynamohum yes i got it working again. i did the steps i mentioned in that MS Community post i linked above. with respect to my own PC set-up at least, basically it seems like any advice to uninstall anything is wrong and that i (and any users at all?) should just go ahead and download this github trackpad driver (the one we're talking about here) and install it. honestly i don't know how to uninstall it once it is installed, it's got something to do with locating its files in one of those Windows folders that are stashed away somewhere in File Explorer where app files are generally sent to by default.

i made a note folder for myself detailing the problem i had and how i sort-of fixed it, so that if it ever happens again i'd be better off. you're welcome to look at it if you like, i could upload it here i guess; it's just a text document together with some pdfs of the webpages i link in the note. there was one other MS Community post i made that helped a bit with the problem which i link in that note also... 

as a last resort wouldn't a factory-reset work, though extremely tedious? i had first installed the trackpad driver onto a brand-new laptop with Windows 11 freshly installed and had no problems with it until i tried to mess around with the default trackpad drivers on Windows 11, following the seemingly erroneous advice on the github page to "uninstall the default drivers first[?]". my pc is a dell latitude 7430 laptop with an i7 cpu (i think?) and 16gb ram running windows 11; but i really couldn't say which specs would be relevant to getting a driver for a trackpad to work...

one little problem i still have with it is that whenever i start-up my pc i need to "remove" the trackpad from my bluetooth devices and then pair it again before it my pc will recognise it. but i can't really complain about that.

i did e-mail the developer, "imbushuo", with questions but they never replied; but i can't resent them too much for that as the driver is free after-all and is really good once it's working. 

you installed it once successfully right? i remember i had some initial difficulty doing that because of some security feature in the BIOS (the pre-boot operating-system?) that prevented me from installing third-party drivers that i had to learn how to bypass through some google help. 

i wish you good luck with it because it really is good once it's working and it would probably take an age to figure out how to code a driver for it myself!