imbushuo/mac-precision-touchpad #528: Magic Trackpad battery level display

zhaowenjunan, 2023-11-14:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
I'm always frustrated when the trackpad suddenly ran out of power.

**Describe the solution you'd like**
Is it possible to display the power level of the Magic Trackpad at the system level.

**Describe alternatives you've considered**
Bluetooth device battery level display specification.

lloydo3000, 2023-11-15:
Duplicate of https://github.com/imbushuo/mac-precision-touchpad/issues/413

vip991988, 2024-02-15:
please update Magic Trackpad battery level display. Thank you so much !!!