imbushuo/mac-precision-touchpad #529: Will this work with the ORIGINAL Magic Trackpad 1?

srgzx, 2023-11-23:
It doesn't seem to on my computer.  Can anyone offer advice to get the MT1 working?

srgzx, 2024-01-16:
I gave up. The MT1 is kind of crappy anyway.  Only decent thing about it is the removable battery.

dmitry-entin, 2026-10-29:
Draft PR with Magic Trackpad 1 (Bluetooth) support: #579

Heads-up: it was written with an AI assistant and tested on one computer only (Windows 11 x64, MT1 over Bluetooth). Cursor, tap, scroll, pinch to zoom and 3/4-finger gestures work there. It is not officially signed, so trying it needs a self-built driver and test signing enabled (which requires Secure Boot off).

If you try it, please report your results in the PR, especially on other hardware.