imbushuo/mac-precision-touchpad #530: Drivers Can't work at Win 11 23H2

sinfee, 2023-11-25:
In Win11 23H2 , After the driver ver 3979 is installed and connected to the trackpad, it will automatically disconnect after the connection is successful, and then device is deleted

crownAzaf, 2023-11-29:
This project is just dead. Do you guys know any alternatives? I wouldn't mind paying for it as well

crownAzaf, 2023-12-13:
> The [Magic Utilities](https://magicutilities.net) are alive since 12 years.

they don't offer the same gesture support as windows 11 precision drivers. Right now I have 3 fingers and 4 fingers gestures as well as taps working but everything is very buggy.

DmitriyRomanovich, 2023-12-25:
all ok 
Iam  just testing on win 11
I insaled via choco

RlxPudding, 2023-12-25:
I istalled it also with choco on win11 it works like acspacted. About 15 days ago and I'm happy, it feels like the magic Trackpad just made for the win11, because of that thank you very much for this solution.

jonprocter, 2023-12-30:
> This project is just dead. Do you guys know any alternatives? I wouldn't mind paying for it as well

The official Apple driver works great (it's not just for T2 machines, works with any Windows computer) - the install process is a little more involved though. You have to

1. Download this: https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe 
2. Extract it with 7Zip (right click > extract) because the exe can only run on Apple computers
3. Go to the ApplePrecisionTrackpadBluetooth folder, right click the .inf and install (and repeat for the USB folder)

fitchmultz, 2024-01-03:
> > This project is just dead. Do you guys know any alternatives? I wouldn't mind paying for it as well
> 
> The official Apple driver works great (it's not just for T2 machines, works with any Windows computer) - the install process is a little more involved though. You have to:
> 
> 1. Download this (the Bootcamp update that had the Precision driver in): https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe
> 2. Extract it with 7Zip (right click > extract) because the exe can only run on Apple computers
> 3. Go to the ApplePrecisionTrackpadBluetooth folder, right click the .inf and install (and repeat for the USB folder)

This worked for me. I installed both for good measure.

clemrdh, 2024-01-05:
> The official Apple driver works great (it's not just for T2 machines, works with any Windows computer) - the install process is a little more involved though. You have to:
>
>     1. Download this (the [...]

Thanks a lot, worked for me as well !

Jiaqin-Tan, 2024-03-05:
> The official Apple driver works great (it's not just for T2 machines, works with any Windows computer) - the install process is a little more involved though. You have to:
>
> 1. Download this (the [...]

Thank you so much for teaching us!

jonprocter, 2024-03-05:
> Thank you so much for teaching us!

No problem!

erikmartinessanches, 2024-06-28:
> The official Apple driver works great (it's not just for T2 machines, works with any Windows computer) - the install process is a little more involved though. You have to:
>
> 1. Download this (the [...]

I am confused where the steps are supposed to be done, in the macOS site or Windows?

jonprocter, 2024-06-28:
> I am confused where the steps are supposed to be done, on the macOS side or Windows?

Windows 🙂

BarakXYZ, 2024-10-21:
Here on WIN 11 -> This package by imbushuo works really well!
Easily install via "choco: choco install mac-precision-touchpad" (without quotes)

haven't tested the officially supported drivers mentioned in this thread as I can't retrace their original download link from Apple and kinda afraid to download these drivers from a random link. But it looks promising too!
If any one can find the original link from Apple to download these drivers that would be cool.

Thanks!

jonprocter, 2024-10-21:
> haven't tested the officially supported drivers mentioned in this thread as I can't retrace their original download link from Apple and kinda afraid to download these drivers from a random link. But [...]

It was a nightmare to find, and I can't actually remember how I did it sorry. It's a file intended to be downloaded by the Apple Software Update tool and doesn't have a link on support.apple.com etc. It is the original link though, it's hosted on apple.com.

BarakXYZ, 2024-10-21:
> It was a nightmare to find, and I can't actually remember how I did it sorry. It's a file intended to be downloaded by the Apple Software Update tool and doesn't have a link on support.apple.com [...]

Gotcha. Cool, thanks!

jonprocter, 2024-10-21:
> Gotcha. Cool, thanks!

Oooo I just googled and I think this was how I found it: https://www.reddit.com/r/Leebal/comments/yoadoo/guide_install_latest_official_magic_trackpad_on/

BarakXYZ, 2024-10-21:
> > Gotcha. Cool, thanks!
> 
> Oooo I just googled and I think this was how I found it: https://www.reddit.com/r/Leebal/comments/yoadoo/guide_install_latest_official_magic_trackpad_on/

Awesome, Thanks for the efforts, appreciate it!

maashrafh, 2024-12-07:
Sorry for necro-ing this issue; I just wanted to let you all know that I've made a fork of a more recently updated version of this project, which can be found here: https://github.com/maashrafh/MagicTrackpad2ForWindows

There is some slight change in how the driver is installed due to signing issues, especially if you are using the USB C Magic Trackpad (afaik Magic Trackpad 2/3 is unaffected, since the .inf file is identical to this project).

Let me know if you run into issues with my fork.

rin67630, 2025-07-19:
> The official Apple driver works great (it's not just for T2 machines, works with any Windows computer) - the install process is a little more involved though. You have to:
>
> 1. Download this (the [...]

That appears to work for Appe's external Magic Trackpads, not for the trackpad built in Macbooks (not connected via USB nor Bluettooth, but over SPI. Right?

stephen147, 2025-10-05:
I followed the reddit thread there, the only thing extra that was needed, was to do a restart.

Does anyone know the gestures after installing **ApplePrecisionTrackpadBluetooth.inf**?

Is there a way to customise them?