imbushuo/mac-precision-touchpad #537: No touchpad option on Win 10

ralphling, 2024-04-04:
****Describe the bug****
There is no "touchpad" settings options in system settings. It seems to be missing.

****Environment****
Running Win 10
64-bit operating system, x64-based processor
OS Build 19045.4170
Version 22H2
Edition Windows 10 Home

JensCarmbrant, 2024-06-20:
I'm sorry to be one of those "I have the same problem", but I do, so here I go.
This package works great on my Windows laptop as it has a built in trackpad. But on my Windows Desktop machine, there is no touchpad control settings. I have scoured the internet for any way to "add" this option when the computer does not have a native touchpad, but so far no luck.

If someone else has gotten this to work on a Windows desktop, please reach out!

Thanks

gch90, 2025-01-28:
I have the same exact issue where I cannot summon the trackpad settings in windows 11 desktop

yinsp398, 2025-02-02:
I have the same issue too, so I have replaced HUKE Trackpad with TouchPad. 
You can found cheaper link with JD(https://item.jd.com/10093709270680.html)