imbushuo/mac-precision-touchpad #539: I would like to get the HID report descriptor for the Mac touchpad.

lhyyydsnbb, 2024-07-03:
Hello,

I have reviewed the source code on your GitHub. I would like to extract the HID report descriptor for the Mac-compatible touchpad used in this project. I intend to use it in microcontroller development to create a touchpad similar to Apple's. However, the project contains a large amount of code, and I am unsure which report descriptor is for the Mac-compatible touchpad. Could you please reply when you see this message and ideally provide me with information on how many bytes of data this report descriptor should send and what each byte represents? Thank you very much.

Best regards,

Letorshillen, 2024-10-29:
Hey, I saw that you didn't get an answer yet.
Did you maybe find a report descriptor for the mac touchpad as I am also looking for a working descriptor.
Apple has an example descriptor [here](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf#//apple_ref/doc/uid/DontLinkBookID_1-BK99CH25-BK99CH25SLK0292).
But when using this descriptor to create a HID-Device i cannot get it to work.

lhyyydsnbb, 2024-10-30:
> 嘿，我看到你还没有得到答案。您是否找到了 mac 触摸板的报告描述符，因为我也在寻找一个有效的描述符。 Apple [在此处](https://developer.apple.com/accessories/Accessory-Design-Guidelines.pdf#//apple_ref/doc/uid/DontLinkBookID_1-BK99CH25-BK99CH25SLK0292)提供了一个示例描述符。但是当使用此描述符创建 HID-Device 时，我无法让它工作。

no, I'm abandon this idea