imbushuo/mac-precision-touchpad #542: Right Click and Drag

gogozhe, 2024-07-18:
**Describe the bug**
I tend to play Roblox, yes you heard that right, with the magic trackpad, in this game it requires you to right click, and drag to move around your camera, however these drivers do not support click and drag with the right click. I know this isn't only a problem with a game cause I've tried on other things such as just my windows screen background, you cannot right click and drag. These drivers are flawless imo other than the right click and drag.

sullerandras, 2024-11-17:
I have the same issue with other games. Any chance to fix it?

gogozhe, 2026-09-05:
I've been using Magic Utilities to fix this exact issue and many others, a great program for using Apple devices on Windows, there is definitely a way to rewrite these drivers by vibe coding it with Fable 5.2 or GPT Astra 6