imbushuo/mac-precision-touchpad #548: Kindly Request for Support for the New USB-C Touchpad

EachSheep, 2024-10-30:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
We sincerely request support for the newly released touchpad with a USB-C interface, launched on October 28, 2024.

Hello, I am a user of both Windows and macOS. On October 28, 2024, I purchased Apple's newly released touchpad with a USB-C interface.
However, I noticed that this new touchpad does not seem to be supported by the Release 3979 driver. On the same machine, the older version of the touchpad with a Lightning interface is well-supported.

My system is Windows 11 Pro 23H3.

**Describe the solution you'd like**
Add support for the new touchpad with USB-cC interface.

CaptPickguard, 2024-10-30:
I too have run into this issue. It appears the official Bootcamp drivers also are unable to properly make use of the new device... unless there is a new release from Apple I am not aware of. Hoping support can be added to this driver!

StephenWatson6, 2024-11-04:
USBC新版妙控板：硬件不变，蓝牙3.0升级至5.2。不支持15.0以下macos系统。
不适配该windows驱动，没有触摸板设置选项，无法使用手势功能，有线连接时报驱动错误。
拆解评测视频：https://www.bilibili.com/video/BV1snSmYBE5e/

StephenWatson6, 2024-11-04:
建议继续用lighting版，不推荐USBC版

StephenWatson6, 2024-11-04:
> 建议继续用lighting版，不推荐USBC版

硬件不变，兼容性倒退，不支持15.0以下macos系统。
不适配该windows驱动，无触摸板设置选项，无手势功能，有线连接时报驱动错误。
拆解评测视频：https://www.bilibili.com/video/BV1snSmYBE5e/

noahhhi, 2024-11-06:
The new trackpad seem does not support multi-touch on Windows, including iPad and Mac (unless you upgrade to latest system).

CaptPickguard, 2024-11-06:
What I wonder is does a new version of bootcamp support it... I wonder if
someone could pull the driver from a newer Mac if so.

takahar, 2024-11-06:
Does someone check the USB Device ID of the USB-C Magic Trackpad? 
(ref: https://kb.synology.com/ja-jp/DSM/tutorial/How_do_I_check_the_PID_VID_of_my_USB_device )

In many cases, simply adding the USB Device ID to the driver INF file and reinstalling the driver resolves compatibility issues with the newer model of the device.

FoldableChair, 2024-11-06:
> Does someone check the USB Device ID of the USB-C Magic Trackpad? (ref: https://kb.synology.com/ja-jp/DSM/tutorial/How_do_I_check_the_PID_VID_of_my_USB_device )
>
> In many cases, simply adding the USB [...]

I think i found the device id info for the usbc touchpad (VID_05AC&PID_0324) and i tried replacing it in the driver but i got a hash catalog file error and i have no idea how to fix that (im a complete amateur)

398noe, 2024-11-06:
As far as I checked, the hardware ID was shown as follows `USB\VID_05AC&PID_0324&REV_0410&MI_00`.

However, when I check the source code, in Device.c of the AmtPtpHidFilter project, it works like sending STATUS_NOT_SUPPORTED to TraceEvents if it is not a VID / PID of TRACKPAD_2.

https://github.com/imbushuo/mac-precision-touchpad/blob/29d277880f167a3cead6c1fe8b4eba58bc168a05/src/AmtPtpHidFilter/Device.c#L306-L317

This may indicate that simply appending the pid and vid to the inf file will not work.

mlo0352, 2024-11-18:
I saw that nobody can make releases because they're unable to sign - would there by any way for us to make changes and attempt to create an unsigned driver? I have never worked with drivers, but I'm more than willing to help modify the code to work. 

The MT3 working is very ideal for people, who, like myself, need to have all of their peripherals working on USB-C.

airtaxi, 2024-11-27:
I don't have this product but if you follow the instruction of this [link](https://pureinfotech.com/disable-driver-signature-enforcement-windows-11/), you can install any unsigned drivers.

mlo0352, 2024-11-27:
> > 
> 
> well magic utilities is able to get it working so im gonna assume its possible
> 
> Edit: has anyone tried contacting the developer directly?

Also, and maybe I'm doing something wrong here, Magic Utilities doesn't have full features - they're missing three finger swipe-up and swipe-to-side and pinching (although they claim this is because there's no Windows equivalent to the pinch-to-zoom action).

Gentoli, 2024-12-09:
I have a diff that adds the new USB device ID (will add the Bluetooth ones when I got to using it). All the precision features seems to be working for me.

https://github.com/Gentoli/magic-trackpad-precision/commit/2f9b21821ec9fa5ae48de1a03b456b6a82798203

On publishing / creating release without EV, I know [Virtual-Display-Driver](https://github.com/VirtualDisplay/Virtual-Display-Driver) uses a self-signed testing cert. I'm not sure how will that interacts with secure boot.

CodingOX, 2024-12-17:
This method should support your USB-C device
---
1. Download this (the Bootcamp update that had the Precision driver in): [[swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe](https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe)](https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe)

2. Extract it with 7Zip (right click > extract) because the exe can only run on Apple computers

3. Go to the ApplePrecisionTrackpadBluetooth folder, right click the .inf and install (and repeat for the USB folder)

CaptPickguard, 2024-12-17:
> ## This method should support your USB-C device
>
>     1. Download this (the Bootcamp update that had the Precision driver in): [...]

This is an old driver from before the USB-C launch. Someone did an update and wiresharked to get the direct link. If someone could do whatever they did again to get the actual latest Bootcamp drivers we may have success, but these older drivers I can confirm do not work.

mlo0352, 2024-12-17:
> 1. Download this (the Bootcamp update that had the Precision driver in): [[swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe](https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe)](https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe) <http://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe%5D(https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe)%5D(https://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe)>

What are you referring to with regard to the wiresharking?

CaptPickguard, 2024-12-17:
Someone pulled that apple link by grabbing packages from their macbook
installing bootcamp drivers. It's been floating around for a while.

vitoplantamura, 2024-12-20:
hi Everyone,

In my fork of the imbushuo driver, we are trying to support the USB-C Magic Trackpad 2 (no bluetooth):

https://github.com/vitoplantamura/MagicTrackpad2ForWindows

For the installation we are testing different solutions, in an attempt to avoid having to buy a code signing certificate.

If you have a USB-C MT2, every tester is welcome! (especially if you have access to a PC with Windows on ARM).

More info here: https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/13

Thanks, Vito

martinlaven, 2025-01-07:
Hi,

I had the same problem and I want to use the official drivers. It seems like the latest one from: http://swcatalog.apple.com/content/catalogs/others/index-windows-1.sucatalog is still the old driver from 2022. 

I followed the link from vitoplantamura and followed his instructions, but I used the official 2022 drivers and it seems to be working fine :-)  So thank you very much vitoplantamura (and jbsilva)!

Best regards,
Martin

mlo0352, 2025-01-07:
To clarify, this is just for wired use, correct?

martinlaven, 2025-01-07:
> To clarify, this is just for wired use, correct?
> […](#)

Correct. I wouldn't be suprised if it works to do the same thing with bluetooth as well, but I never use it so I don't have the time to check it up. But if you find the correct device then I think it will be possible to just use the bluetooth driver.

lc700x, 2025-03-14:
> I think i found the device id info for the usbc touchpad (VID_05AC&PID_0324) and i tried replacing it in the driver but i got a hash catalog file error and i have no idea how to fix that (im a [...]

Have you tried this @FoldableChair to install the modified driver?

1. **Disable Driver Signature Enforcement**:
   - Restart your computer and enter the boot menu by pressing `Shift` while selecting "Restart."
   - Navigate to `Troubleshoot > Advanced Options > Startup Settings` and click "Restart."
   - When the options appear, press the key for "Disable driver signature enforcement" (usually `F7`).

2. **Install the Unsigned Driver**:
   - Locate your driver file (e.g., `.inf`) and right-click on it.
   - Select "Install" and proceed with any warnings.

rocknroLLa89, 2025-03-15:
Hi Everyone, I just was gifted the usb c magic trackpad and it did connect to windows, but cant use the scroll or gestures or multifinger functionalities, I am reading here that few have found it successful over Bluetooth, can someone please explain to me how?, PS I am green as they come so please consider that, Thank you

sjain882, 2025-07-06:
> Follow this guide. It works perfectly on the new USB-C touchpad with Bluetooth [vitoplantamura/MagicTrackpad2ForWindows#20](https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/20)

Looks like someone has made a YouTube guide for this, as well as an alternative set of instructions: https://github.com/imbushuo/mac-precision-touchpad/issues/556#issuecomment-2815869915