imbushuo/mac-precision-touchpad #556: can you make the version drivers for magic trackpad USB Type - C

harsh-uiux, 2025-02-19:
The latest release of the newest magic trackpad is not working. I need to get it working. would be a god help if you could create a driver update for the type c magic trackpad.

AnshulJ999, 2025-02-24:
https://github.com/vitoplantamura/MagicTrackpad2ForWindows

Check this for type-c support (USB only, no Bluetooth)

Charminoy, 2025-02-27:
True, the driver is not working for the A3120 model (the type-C one), sad.

harsh-uiux, 2025-03-01:
> https://github.com/vitoplantamura/MagicTrackpad2ForWindows
> 
> Check this for type-c support (USB only, no Bluetooth)

I did try this but the gestures dont work at all.

AnshulJ999, 2025-03-01:
> > https://github.com/vitoplantamura/MagicTrackpad2ForWindows
> > Check this for type-c support (USB only, no Bluetooth)
> 
> I did try this but the gestures dont work at all.

Just want to ask if you had done these extra steps for the USB C version? 

[image]

Since the hardware IDs and stuff are different, there's some issues. But you could try raising an issue on that repo and sharing details. May be able to solve it. 

AFAIK it's the only known solution for USB-C version, and also the best driver for older lightning models too.

davidfi01, 2025-03-13:
Is there any work around for the USB-C version to make it work over bluetooth as well?  All I can find are references to the usb driver not being ported for the new trackpad yet.

PaulShiLi, 2025-03-14:
> Is there any work around for the USB-C version to make it work over bluetooth as well? All I can find are references to the usb driver not being ported for the new trackpad yet.

Unfortunately, there's no other way as far as I know besides using apple's bluetooth drivers which worked for me.
Try giving this a shot by extracting AppleBcInstaller.exe and installing the driver from the extracted folder:
https://www.reddit.com/r/bootcamp/comments/ygv1mh/comment/jcn4rn1/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

PaulShiLi, 2025-03-18:
> > > Is there any work around for the USB-C version to make it work over bluetooth as well? All I can find are references to the usb driver not being ported for the new trackpad yet.
> > 
> > 
> > Unfortunately, there's no other way as far as I know besides using apple's bluetooth drivers which worked for me. Try giving this a shot by extracting AppleBcUpdate.exe and installing the driver from the extracted folder: https://www.reddit.com/r/bootcamp/comments/ygv1mh/comment/jcn4rn1/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button
> > AppleBcUpdate link: http://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe
> > Gestures and everything else works as well
> 
> Did you really managed to connect your USB-C trackpad version via bluetooth on Windows? Because I tried and the driver you link is from 2022. It's also not accepted by Windows, and says the driver is not compatible with this device.

Hmm, strangely it works on my end! (You will have to manually update the driver called Bluetooth HID or something similar that starts with bluetooth under the HID section in Device Manager with Apple's Bluetooth Drivers). Once you've mapped it, you should see Apple Bluetooth Precision Trackpad. 

I'm currently running on x64 based windows so I'm not sure if this works on x32 or ARM based Windows. 

[image]

[image]

artik, 2025-04-13:
Oh woaw, it works! Fantastic, thank you very much :)

LincolnWasden, 2025-04-18:
For anyone still not able to get it to work, here is a youtube video displaying these steps, step by step:
https://www.youtube.com/watch?v=J_z9-rE_6zI

If you do not like videos: 

Here are the written steps: (side note this will give you scrolling and gestures but only via bluetooth)
1. Connect Apple Trackpad (USBC) via Bluetooth
2. Go to Device Manager -> "Human Interface Devices" -> Locate "Bluetooth HID Device" (if you cant find it, see if under "Human Interface Devices" you see "Trackpad" or "Apple Trackpad" if so, click the red x and uninstall the device. This now repeat step 1 and connect your device via bluetooth)
3. Now that your device displays as "Bluetooth HID Device". Go to chrome and download 7 ZIP (if you dont already have it installed)
4. Now copy this link and paste it in your URL box and hit enter to download: http://swcdn.apple.com/content/downloads/03/60/041-96205/61hhcnj7q5dxosc171ytixty20vuqg0r0n/AppleBcUpdate.exe
5. Now go to downloads and right click on "AppleBcUpdate" and click "show more options" and then look for 7 Zip -> hover your mouse over 7Zip and then select "Extract Here" 
6. Open the ApplePrecisionTrackpadBluetooth folder and then right click on the INF file "ApplePrecisionTrackpadBluetooth" (in the Type column of file explorer it should be the file with a type of "Setup Information") Then click "Install"
7. After you installed the "ApplePrecisionTrackpadBluetooth" INF file -> Device Manager -> Bluetooth HID Device -> Right Click on Bluetooth HID Device -> Update Driver -> Browse My Computer for Drivers -> Let me Pick from a list of available drivers on my computer -> Click "Have a disk" -> Select the "ApplePrecisionTrackpadBluetooth" INF file -> Open -> Okay -> Next -> Yes
8. **Optional** You may need to reconnect your trackpad via bluetooth
9. Congrats! You're done! and scrolling and gestures (*some*) now work on Windows!!

sjain882, 2025-07-06:
Someone else also got BT working with MT2 USB-C and provided some more details here (such as driver version): https://github.com/vitoplantamura/MagicTrackpad2ForWindows/issues/20

sjain882, 2025-07-06:
Would be nice if someone could confirm their model number please.

The MT2 Lightning, MT2 USB-C, MT3, MT4 exists, but I'm not sure which one this issue refers to.

Charminoy, 2025-07-06:
Mine is the MT USB-C one, which is also called MT4. There is only one model
for the USB-C Trackpad.